pymodbus-dev/pymodbus #1528: RTU framer fails on packaging a readWrite request function code 0x17 (23)

erikvanhamme, 2023-04-26:
### Versions

- Python: 3.10.1
- OS: Windows 10
- Pymodbus: 3.2.2
- Modbus Hardware (if used):

### Pymodbus Specific

- Client: rtu - sync

### Description

Trying to use the combined readWrite call for holding registers. I get error from the internals when the request is being packaged.

The first read call works, as indicated by 
[266, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
in the output.

The readWrite call always fails.

### Code and Logs

```python

from pymodbus.client import ModbusSerialClient
import pymodbus

pymodbus.pymodbus_apply_logging_config()

client = ModbusSerialClient(method="rtu", port="COM6", stopbits=1, bytesize=8, parity='N', baudrate=115200, timeout=1)

client.connect()

read = client.read_holding_registers(0, 32, 1)
print(read.registers)

print(client.readwrite_registers(read_address=0, read_count=32, write_address=1, values=[0], slave=1))

client.close()

```

Output:

```
2023-04-26 16:48:59,398 DEBUG logging:102 Current transaction state - IDLE
2023-04-26 16:48:59,398 DEBUG logging:102 Running transaction 1
2023-04-26 16:48:59,399 DEBUG logging:102 SEND: 0x1 0x3 0x0 0x0 0x0 0x20 0x44 0x12
2023-04-26 16:48:59,399 DEBUG logging:102 New Transaction state "SENDING"
2023-04-26 16:48:59,399 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-04-26 16:48:59,483 DEBUG logging:102 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-04-26 16:48:59,484 DEBUG logging:102 RECV: 0x1 0x3 0x40 0x1 0xa 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x82 0x1a
2023-04-26 16:48:59,485 DEBUG logging:102 Getting Frame - 0x3 0x40 0x1 0xa 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-04-26 16:48:59,486 DEBUG logging:102 Factory Response[ReadHoldingRegistersResponse': 3]
2023-04-26 16:48:59,487 DEBUG logging:102 Frame advanced, resetting header!!
2023-04-26 16:48:59,487 DEBUG logging:102 Adding transaction 1
2023-04-26 16:48:59,488 DEBUG logging:102 Frame - [b'\x01\x03@\x01\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x82\x1a'] not ready
2023-04-26 16:48:59,489 DEBUG logging:102 Getting transaction 1
2023-04-26 16:48:59,489 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
[266, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-04-26 16:48:59,489 DEBUG logging:102 Current transaction state - TRANSACTION_COMPLETE
2023-04-26 16:48:59,489 DEBUG logging:102 Running transaction 2
Traceback (most recent call last):
  File "C:\Users\EVanHa\projects\azure\AGL-Power\fw-gtccr-control\test_readwrite.py", line 13, in <module>
    print(client.readwrite_registers(read_address=0, read_count=32, write_address=1, values=[0], slave=1))
  File "C:\Toolchain\Python\Python310\lib\site-packages\pymodbus\client\mixin.py", line 470, in readwrite_registers
    return self.execute(
  File "C:\Toolchain\Python\Python310\lib\site-packages\pymodbus\client\base.py", line 198, in execute
    return self.transaction.execute(request)
  File "C:\Toolchain\Python\Python310\lib\site-packages\pymodbus\transaction.py", line 170, in execute
    response, last_exception = self._transact(
  File "C:\Toolchain\Python\Python310\lib\site-packages\pymodbus\transaction.py", line 284, in _transact
    packet = self.client.framer.buildPacket(packet)
  File "C:\Toolchain\Python\Python310\lib\site-packages\pymodbus\framer\rtu_framer.py", line 242, in buildPacket
    data = message.encode()
  File "C:\Toolchain\Python\Python310\lib\site-packages\pymodbus\register_read_message.py", line 289, in encode
    result += struct.pack(">H", register)
struct.error: required argument is not an integer
```

janiversen, 2023-04-26:
we have some fixes for read write on dev, that might help you.

janiversen, 2023-04-27:
Just saw a problem with your code (not causing the problem you reported), you are using address=0, that is normally not allowed in modbus devices.

erikvanhamme, 2023-04-27:
$ pip list | grep pymodbus
pymodbus                      3.3.0a0     c:\users\evanha\projects\pymodbus

Dev version did indeed solve my problem, thank you.

closed 2023-04-27 by erikvanhamme