pymodbus-dev/pymodbus #1529: Offset registers > 100 responding IllegalAddress in TCP server mode

neosoyo, 2023-04-26:
### Versions

- Python: 3.10.6
- OS: Pop!_OS 22.04 LTS
- Pymodbus: 3.2.2

### Pymodbus Specific

- Server: tcp - async

### Description

Simulating devices with sparse DataBlock using offsets > 100 is reporting Illegal Address.

### Code and Logs

Config of server:
```python
sc501_datablock = ModbusSparseDataBlock({
            20000: [45, 241, 48],       #ACDC Output current, input voltage and output voltage
            20008: 38,                  #Ambient temperature ACDC
            20023: [0, 0, 0, 0, 0, 0],  #Alarms rectifier
            22001: 23,                  #Output current DCDC
            22003: 48,                  #Output Voltage DCDC
            30052: [4, 2, 1],           #Alarms counts
            48030: [0, 0],              #LVD alarm state
            48040: [0, 0],              #batt + fuse alarm
            48050: [0, 0],              #batt - fuse alarm
            48060: [0, 0],              #batt alarm
            48140: [0x4208,0xcccd],     #Ambient temperature alarm
            48146: [0x42ad, 0x0000],    #Humidity alarm 
            48170: [0,0,0,0,0,0],         #Alarms type and classication
            48190: [0,0],               #BMS alarm
            48200: [0,0],               #ACDC rectifier alarm
            48204: [0,0],               #Solar converter alarm
            48216: [0,0],                #DCDC rectifier alarm
        })
Battery_datablock = ModbusSparseDataBlock({
            1: 6720,
            2: 130,
            3: [4200,4200,4200,4200,4200,4200,4200,4200,4200,4200,4200,4200,4200,4200,4200,4200],
            19: [39,40,43,2100,170,69,80,2,0,0,0],
            30: [0x0D,0xFE],
            32: [0,0,32,32],
            37: [16,3920],
            62: [0,0,0,0,0,0,0,0,0,0,0],
            81: [0,0,0,0],
            91: [0,0,0,0,0,0,0,0,0,0,0],
            105: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24],
            117: [56, 30, 33, 52, 30, 34]
        })
context = {
        0x01: ModbusSlaveContext(
            hr=sc501_datablock,
        ),
        0x02: ModbusSlaveContext(
            hr=Battery_datablock,
        ),}
```
My code after requesting all some holding registers, all offset < 100 works fine:
```
battery reg 105, 24 values, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 20000, 3 values, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 20008, 1 value, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 20023, 6 values, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 22001, 1 value, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 22003, 1 value, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 30052, 3 values, modbus error
Exception Response(131, 3, IllegalAddress)
SC501 reg 48170, 6 values, modbus error
Exception Response(131, 3, IllegalAddress)
```
Server response:
```
2023-04-26 17:43:51,271 INFO  logging:96 Server(TCP) listening.
2023-04-26 17:43:59,223 ERROR logging:114 Exception response Exception Response(131, 3, IllegalAddress)
2023-04-26 17:43:59,223 ERROR logging:114 Exception response Exception Response(131, 3, IllegalAddress)
2023-04-26 17:43:59,224 ERROR logging:114 Exception response Exception Response(131, 3, IllegalAddress)
2023-04-26 17:43:59,224 ERROR logging:114 Exception response Exception Response(131, 3, IllegalAddress)
2023-04-26 17:43:59,224 ERROR logging:114 Exception response Exception Response(131, 3, IllegalAddress)
2023-04-26 17:43:59,224 ERROR logging:114 Exception response Exception Response(131, 3, IllegalAddress)
```
Anyone have an idea how to deal with?

janiversen, 2023-04-27:
Since you are not sharing parts of the relevant code, it is quite hard to tell.

However this could very well be caused by a wrong or missing slave_id in the client call, or in the server setup.

janiversen, 2023-04-27:
You say that <100 works fine (no log supplied) so it cannot be the slave_id unless you use different calls.

janiversen, 2023-04-27:
It seems to be a problem in your code, just added a test case that tests SparseDatablock with your setup.

closed 2023-04-27 by neosoyo