pymodbus-dev/pymodbus #1531: Not possible to decode then encode a DiagnosticStatusResponse

JesperHakansson, 2023-04-27:
### Versions

- Python: 3.8.6
- OS: Win10
- Pymodbus: 3.2.2
- Modbus Hardware (if used):

### Pymodbus Specific

- Client: tcp

### Description

It is not possible to decode, then encode a DiagnosticStatusResponse since self.message becomes a tuple in decode and in encode a tuple isn't handled (only list). That means the message will not be processed in encode.

### Code and Logs

import pymodbus.diag_message

a = pymodbus.diag_message.DiagnosticStatusResponse()
a.decode(b"\x00\x01\x02\x03")
packet = a.encode()
print(packet)

Output: b'\x00\x01'

janiversen, 2023-04-27:
Our test "test_diagnostic_requests_decode" does not seem to have any problems. 

Why do you do both encode() and decode(), our classes are not prepared, or at least tested for that. You normally only do either decode() or encode() depending whether you want to send or receive a message.

But pull requests are welcome.

JesperHakansson, 2023-04-27:
Hi Jan,
Thank you for your quick response. 
It was only to visualize the problem that I did encode/decode like that. The problem I have is that in a response I get (response = master.execute(request)) the message attribute is a tuple so when I encode the message it isn't handled correct (is skipped) since the encode method doesn't handle tuples. However, I think a decode/encode should give you back the the same as your input like in my example above?
There doesn't seem to be any test for response_encode (only requests_encode). It also seems like only the function code is verified, not the message, in test_diagnostic_response_decode (line 152).
Also, self.responses doesn't seem to be used anywhere in the test and line 102 and 103, that specifies DiagnosticStatusResponse and DiagnosticStatusSimpleResponse is commented out.
Best regards Jesper

closed 2023-04-27 by janiversen

janiversen, 2023-04-27:
Solved, see dev.

JesperHakansson, 2023-04-27:
Thank you very much!

PR #1533 (2023-04-27): DiagnosticStatus encode missing tuple check.

fixes #1531