pymodbus-dev/pymodbus #1536: AsyncModbusSerialClient aborting on close and crashing

Original-Nativy, 2023-05-04:
### Versions

- Python: 3.10.6
- OS: MacOS 13.1.3 Ventura
- Pymodbus: 3.2.2
- Pyserial: 3.5
- Modbus Hardware (if used): RS485

### Pymodbus Specific

- Client: serial - async

### Description

I am trying to open async serial client, do some communication and close it. During the closing abort gets called with close and _call_connection_lost gets called twice and self._serial.flush() get called on self._serial == None and it crashes

### Code and Logs

```python
import asyncio
from pymodbus.client import AsyncModbusSerialClient

async def read_register(client: AsyncModbusSerialClient, count, address: int):
    result = await client.read_holding_registers(address, count=count, slave=1)
    action = result.registers[99]
    return action

async def read_all_registers(start_address, count, client):
    results = []
    for _ in range(20):
        data = await asyncio.create_task(read_register(client, count,start_address))
        results.append(data)
    return results

async def main():
    client = AsyncModbusSerialClient(port='/dev/tty.wchusbserial10', baudrate=19200, bytesize=8, parity='E', stopbits=2)  
    await client.connect()
    registers = await read_all_registers(1000, 100, client)
    print(registers)
    await client.close()

if __name__ == '__main__':
    asyncio.run(main())
```

```text
Exception in callback SerialTransport._call_connection_lost(None)
handle: <Handle SerialTransport._call_connection_lost(None)>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pymodbus/client/serial_asyncio/__init__.py", line 433, in _call_connection_lost
    self._serial.flush()
AttributeError: 'NoneType' object has no attribute 'flush'
```

janiversen, 2023-05-05:
Hmm seems like a bug, I will take a look.

janiversen, 2023-05-05:
Another user reported this a while ago: https://github.com/pymodbus-dev/pymodbus/discussions/1504

Solved on dev, will be part of version 3.3

closed 2023-05-05 by janiversen