pymodbus-dev/pymodbus #1542: RTU client over TCP response is missing the last byte in the frame buffer

Elfe, 2023-05-21:
### Versions

- Python: 3.11
- OS: Linux (arch)
- Pymodbus: 3.2.2
- Modbus Hardware (if used):

1. PC (192.168.2.2)
2. Arduino with espEasy (192.168.2.247:502) as serial server (tcp)
3. rs485 converter
4. variable frequency drive (ski 780)

### Pymodbus Specific

- Client: tcp/rtu - sync

### Description

When reading registers the response can not be parsed as the response frame in the buffer is missing the last byte from the response.
Data bytes from tcpdump:
01 03 00 02 17 81 2b 9a
From "Current Frame in buffer":
0x1 0x3 0x0 0x2 0x17 0x81 0x2b

Increasing the count to 2 registers extends the data in tcpdump to 10 bytes but pymodbus only shows 9 bytes again.

### Code and Logs

```python
#!/usr/bin/python
import pymodbus
pymodbus.pymodbus_apply_logging_config()

from pymodbus.client import ModbusTcpClient
from pymodbus.transaction import ModbusRtuFramer

c = ModbusTcpClient("192.168.2.247", port=502, framer=ModbusRtuFramer)
success = c.connect()
result  = c.read_holding_registers(address=0x7002, slave=1, count=1)
c.close()
```
Log:
```
2023-05-21 15:02:49,427 DEBUG logging:102 Connection to Modbus server established. Socket ('192.168.2.2', 55012)
2023-05-21 15:02:49,427 DEBUG logging:102 Current transaction state - IDLE
2023-05-21 15:02:49,427 DEBUG logging:102 Running transaction 1
2023-05-21 15:02:49,427 DEBUG logging:102 SEND: 0x1 0x3 0x70 0x2 0x0 0x1 0x3f 0xa
2023-05-21 15:02:49,427 DEBUG logging:102 New Transaction state "SENDING"
2023-05-21 15:02:49,427 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-05-21 15:02:49,540 DEBUG logging:102 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-05-21 15:02:49,540 DEBUG logging:102 RECV: 0x1 0x3 0x0 0x2 0x17 0x81 0x2b
2023-05-21 15:02:49,540 DEBUG logging:102 Frame check failed, ignoring!!
2023-05-21 15:02:49,540 DEBUG logging:102 Resetting frame - Current Frame in buffer - 0x1 0x3 0x0 0x2 0x17 0x81 0x2b
2023-05-21 15:02:49,540 DEBUG logging:102 Getting transaction 1
2023-05-21 15:02:49,540 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

```
tcpdump from the data packets
```
15:02:49.427416 IP 192.168.2.2.55012 > 192.168.2.247.502: Flags [P.], seq 1:9, ack 1, win 64240, length 8
        0x0000:  4500 0030 0d83 4000 4006 a6fb c0a8 0202  E..0..@.@.......
        0x0010:  c0a8 02f7 d6e4 01f6 6b3d 0d4a d97a de40  ........k=.J.z.@
        0x0020:  5018 faf0 866c 0000 0103 7002 0001 3f0a  P....l....p...?.
15:02:49.540400 IP 192.168.2.247.502 > 192.168.2.2.55012: Flags [P.], seq 1:9, ack 9, win 5736, length 8
        0x0000:  4500 0030 b0de 0000 ff06 849f c0a8 02f7  E..0............
        0x0010:  c0a8 0202 01f6 d6e4 d97a de40 6b3d 0d52  .........z.@k=.R
        0x0020:  5018 1668 c5cc 0000 0103 0002 1781 2b9a  P..h..........+.
```

janiversen, 2023-05-21:
Actually the problem are because the response is strange:
```
RECV: 0x1 0x3 0x0 0x2 0x17 0x81 0x2b
```
That is slave=0x01, function code=0x3 and byte count = 0x0.
Due to the byte count being 0x0, CRC is assumed to be 0x2 0x17 which of course leads to a frame check failure.

Assuming byte count is 0x2 (1 register), the frame is legal, because:
register value = 0x2 0x17
crc = 0x81 0x2b

So as far as I can see pymodbus works correctly but your device do not send the correct byte count.

Elfe, 2023-05-21:
Hi,

thanks for the explanation.

After digging around in the manual it appears the device is by default in some "non standard modbus protocol" mode.
After switching that to "Standard Modbus protocol" the byte count is where it should be and the frames decode correctly.

```
DEBUG:pymodbus.logging:Connection to Modbus server established. Socket ('192.168.2.2', 53400)
DEBUG:pymodbus.logging:Current transaction state - IDLE
DEBUG:pymodbus.logging:Running transaction 1
DEBUG:pymodbus.logging:SEND: 0x1 0x3 0x70 0x2 0x0 0x1 0x3f 0xa
DEBUG:pymodbus.logging:New Transaction state "SENDING"
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.logging:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.logging:RECV: 0x1 0x3 0x2 0x17 0x1d 0x77 0xbd
DEBUG:pymodbus.logging:Getting Frame - 0x3 0x2 0x17 0x1d
DEBUG:pymodbus.logging:Factory Response[ReadHoldingRegistersResponse': 3]
DEBUG:pymodbus.logging:Frame advanced, resetting header!!
DEBUG:pymodbus.logging:Adding transaction 1
DEBUG:pymodbus.logging:Frame - [b'\x01\x03\x02\x17\x1dw\xbd'] not ready
DEBUG:pymodbus.logging:Getting transaction 1
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
```

Just for future reference in case someone else stumbles across this issue:
the device is made by Sako
It's a Mini Vector Inverter SKI780
The register controlling the response format is PD-05 the units digit needs to be changed from 0 to 1. The tens digit can stay as is.

Thanks again

closed 2023-05-21 by Elfe