pymodbus-dev/pymodbus #1546: AsyncModbusTcpClient defer_start doesn't seem to work anymore

albireox, 2023-05-24:
### Versions

- Python: 3.10
- OS: macOS 13.4
- Pymodbus: 3.2.2
- Modbus Hardware (if used): N/A

### Pymodbus Specific

- Server: tcp - async
- Client: tcp - async

### Description

It seems the option `defer_start=True` in `AsyncModbusTcpClient` is not honoured anymore in Pymodbus `3.2.2` (I think it was working in 3.1.x). Regardless of whether `defer_start` is `True` or `False`, the function will block once it's awaited. In itself, removing `defer_start`, that's not necessarily a showstopper, and code can be adapted to work with it, but this breaks a lot of code and it's a bit annoying that this breaking change was made in a minor release.

janiversen, 2023-05-24:
If I remember right it also is no longer documented. We do not support defer_start, if you want that you need to instantiate the class e.g. ModbusTcpServer, and do whatever task handling you want.

janiversen, 2023-05-24:
I can see we still have a couple of references, so lets keep this open until all is removed

janiversen, 2023-05-24:
The change is documented in https://github.com/pymodbus-dev/pymodbus/blob/dev/API_changes.rst#version-320
so it should not come as a surprise.

PR #1548 (2023-05-24): Remove references to "defer_start".

fixes #1546
closed 2023-05-24 by janiversen