pymodbus-dev/pymodbus #1549: writing register does not work when running through forwarder

t-dan, 2023-05-24:
### Versions

- Python: 3.11.3
- OS: Arch Linux
- Pymodbus: 3.2.2
- Modbus Hardware (if used): none

### Pymodbus Specific

- Server: tcp - async
- Client: tcp - async

### Description

I am experimenting with writing serial to ETH MODBUS forwarder. I have started from [https://pymodbus.readthedocs.io/en/latest/source/examples.html#serial-forwarder](serial-forwarder) example, modyfying it a bit to forward TCP at one port to another (for testing purposes):

code snippet of the forwarder is:
```c = ModbusTcpClient(host=127.0.0.1', port=target_port)
c = ModbusServerContext(slaves=RemoteSlaveContext(c, slave=0), single=True)

p = int('1'+str(target_port))  # forwarder port, forwards to target_port
s = ModbusTcpServer(c, address=('127.0.0.1', p), allow_reuse_address=True)
s = asyncio.create_task(s.serve_forever())
```

When a client tries to write to the server directly (on target_port), the value is changed and the new value is correctly read.
However, when I try to do the same through forwarder (port f"1{target_port}"), the wrong value is written.

The client looks like:
```client = ModbusClient(host='127.0.0.1', port=port)

rd = client.read_holding_registers(4,count=1)
print('read orig: ',rd.registers)

client.write_register(4,7)  # write new value

rd = client.read_holding_registers(4,count=1)
print('read changed: ',rd.registers)
```

There is ModbusSimulator serving as a server, holding registers are 16bit.

### Code and Logs

Calling twice, going directly to target port, writing 7 when an original value is 5

```
read orig:  [5]
read changed:  [7]
```
```
read orig:  [7]
read changed:  [7]
```

And going through forwarder  to f"1{target_port}", again writing 7 when an original value is 5

```
read orig:  [5]
read changed:  [1]
```
```
read orig:  [1]
read changed:  [1]
```

janiversen, 2023-05-25:
hmmm looks like a bug, let me see if I can reproduce it.

janiversen, 2023-05-25:
Just noted that you are using an example in contrib. As documented contrib contains user contributions that we do not check.

Please try with example modbus_forwarder.py which we have tested.

janiversen, 2023-05-25:
Close as it affects the "as is" examples.

t-dan, 2023-05-25:
I have tried to use `modbus_forwarder.py`, starting it as:

```
python modbus_forwarder.py -p 15020 --client_port 5020 -l debug
```

But it fails with the error listed further. Connecting the slave (modbus simulator) at port 5020 directly works, connecting the forwarder at port 15020 fails. Am I missing something?

The failure of the forwarder is:

```
2023-05-25 22:20:41,595 DEBUG logging:102 Socket [('127.0.0.1', 15020)] opened
2023-05-25 22:20:41,595 DEBUG logging:102 TCP client connection established [('127.0.0.1', 39072)]
2023-05-25 22:20:41,595 DEBUG logging:102 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x4 0x0 0x1
2023-05-25 22:20:41,595 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x4 0x0 0x1
2023-05-25 22:20:41,595 DEBUG logging:102 Factory Request[ReadHoldingRegistersRequest': 3]
2023-05-25 22:20:41,595 DEBUG logging:102 validate[3] 4:1
2023-05-25 22:20:41,596 ERROR logging:114 Datastore unable to fulfill request: 'coroutine' object has no attribute 'isError'; Traceback (most recent call last):
  File "/home/dtihelka/.local/lib/python3.11/site-packages/pymodbus/server/async_io.py", line 267, in execute
    response = request.execute(context)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dtihelka/.local/lib/python3.11/site-packages/pymodbus/register_read_message.py", line 151, in execute
    if not context.validate(self.function_code, self.address, self.count):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dtihelka/.local/lib/python3.11/site-packages/pymodbus/datastore/remote.py", line 50, in validate
    return not self.result.isError()
               ^^^^^^^^^^^^^^^^^^^
AttributeError: 'coroutine' object has no attribute 'isError'

2023-05-25 22:20:41,596 ERROR logging:114 Exception response Exception Response(131, 3, SlaveFailure)
2023-05-25 22:20:41,596 DEBUG logging:102 send: [Exception Response(131, 3, SlaveFailure)]- b'000100000003008304'
2023-05-25 22:20:41,604 DEBUG logging:102 Handler for stream [('127.0.0.1', 39072)] has been canceled
2023-05-25 22:20:41,604 DEBUG logging:102 TCP client disconnected [('127.0.0.1', 39072)]

```

janiversen, 2023-05-26:
hmm that sounds like a missing await, let me try to test.

janiversen, 2023-05-28:
confirmed there is a problem caused by async, I hope we can solve it before version 3.3 is released.

Fredo70, 2023-06-19:
> And going through forwarder to f"1{target_port}", again writing 7 when an original value is 5
> 
> ```
> read orig:  [5]
> read changed:  [1]
> ```
> 
> ```
> read orig:  [1]
> read changed:  [1]
> ```

I think this is fixed in this PR https://github.com/pymodbus-dev/pymodbus/pull/1599

For the other issue, try replacing the AsyncModbusTcpClient with the ModbusTcpClient in the file modbus_forwarder.py. I got it to work that way. But I don't know if it has any side effects.

janiversen, 2023-06-19:
The difference is sync calls which are allowed in a co-routine, nice idea.

There downside is, as far as I can think, only for high-volume usage, because sync blocks and might cause timeouts on the server part.

janiversen, 2023-06-19:
please feel free to submit a pull request that updates the example, so that others can benefit.

janiversen, 2023-06-28:
Copy of #1595

closed 2023-06-28 by janiversen