pymodbus-dev/pymodbus #1551: client.read_coils with simulator fails

tstorek, 2023-05-27:
### Versions

- Python: 3.10 (fresh environment)
- OS: WIN10
- Pymodbus: 3.2.0 and 3.3.0a0
- Modbus Hardware (if used): None

### Pymodbus Specific

- Server: tcp async
- Client: tcp sync/async

### Description

Hello, 
I am new to Modbus and trying to use the new simulator server to develop and test a Modbus to MQTT Gateway. 
I like the idea of having the values changing values during testing due to testing.
Although it seems rather complex to implement own actions like running a simulation. 

Just to make sure: Is the simulator with actions the right way to go? Or would be a datastore with Callbacks the better idea?

Using the example, getting the simulator up and running `examples\server_simulator.py` was pretty easy.
Having the simulator started with the example config, I was able to read the holding registers with sync and async client with `client.read_holding_registers`. This works like a charm for all datatypes, including `float32`. Surprisingly this also reads the "bits" section of the server.
As I understand from the docs (“bits”, defines registers which contain bits (discrete input and coils), this section would configure the coils of the virtual device.
But using, e.g. `client.read_coils(2, 1)` always fails with `Exception Response(129, 1, IllegalAddress)`.

Additionally, I understand the BITS for having boolean values. How can they have int values then?

### Code and Logs

Here is the modified `run_sync_client` from the example. I only added the read commands

```python

def run_sync_client(client, modbus_calls=None):
    """Run sync client."""
    _logger.info("### Client starting")
    client.connect()
    if modbus_calls:
        modbus_calls(client)
    rr = client.read_holding_registers(2, 1)
    print(rr)
    print(rr.registers)
    rr = client.read_coils(2,1)
    print(rr)
    print(rr.bits)
    client.close()
    _logger.info("### End of Program")

if __name__ == "__main__":
    testclient = setup_sync_client(description="Run synchronous client.")
    run_sync_client(testclient)
```

janiversen, 2023-05-27:
The simulator allow you to simulate a complete device, in examples you see an example configuration, and I think in examples/contrib are a couple of extra devices.

Using a datastore (there are no callbacks) is easier but a lot less flexible.

server_async.py is just the servers actions are are in the datastore simulator.

tstorek, 2023-05-28:
@janiversen, thanks for the response. Hence, I am on the right track.
Nevertheless, I need help understanding the BITS section in the example because it says that it defines the coils, but afterward, it is not possible to read them.

When I use `read_coils`, I expect 8 bits returned, including the one I actually want to read.
Unfortunately, the discussion in #1552 does not help here.

janiversen, 2023-05-28:
Of course it support coils and direct input !!!

Please see our documentation about using shared or non-shared blocks.

janiversen, 2023-05-28:
Did you look at our documentation and examples both include coils.

tstorek, 2023-05-28:
@janiversen Yes I did as I initially mentioned.
But starting with the provided example for the simulator, how do I need to change my code above? 
I did not change anything in the simulator example.

janiversen, 2023-05-28:
You need to use the correct coil address at least (see how to calculate address in our documentation).

tstorek, 2023-05-28:
Here is again the config from the provided example:

```python

demo_config = {
    "setup": {
        "co size": 100,
        "di size": 150,
        "hr size": 200,
        "ir size": 250,
        "shared blocks": True,
        "type exception": False,
        "defaults": {
            "value": {
                "bits": 0x0708,
                "uint16": 1,
                "uint32": 45000,
                "float32": 127.4,
                "string": "X",
            },
            "action": {
                "bits": None,
                "uint16": None,
                "uint32": None,
                "float32": None,
                "string": None,
            },
        },
    },
    "invalid": [
        1,
        [6, 6],
    ],
    "write": [
        3,
        [7, 8],
        [16, 18],
        [21, 26],
        [31, 36],
    ],
    "bits": [
        [7, 9],
        {"addr": 2, "value": 0x81},
        {"addr": 3, "value": 17},
        {"addr": 4, "value": 17},
        {"addr": 5, "value": 17},
        {"addr": 10, "value": 0x81},
        {"addr": [11, 12], "value": 0x04342},
        {"addr": 13, "action": "reset"},
        {"addr": 14, "value": 15, "action": "reset"},
        {"addr": 15, "value": True},
        {"addr": 16, "value": True},
    ],
    "uint16": [
        {"addr": [17, 18], "value": 5678},
        {"addr": [19, 20], "value": 14661, "action": "increment"},
    ],
    "uint32": [
        {"addr": [21, 22], "value": 3124},
        {"addr": [23, 26], "value": 5678},
        {"addr": [27, 30], "value": 345000, "action": "increment"},
    ],
    "float32": [
        {"addr": [31, 32], "value": 3124.17},
        {"addr": [33, 36], "value": 5678.19},
        {"addr": [37, 40], "value": 345000.18, "action": "increment"},
    ],
    "string": [
        {"addr": [41, 42], "value": "Str"},
        {"addr": [43, 44], "value": "Strx"},
    ],
    "repeat": [{"addr": [0, 45], "to": [46, 138]}],
}

```
Hence, the first address I can read with this config is equal to 32, as it must be multiples of 16 bits because the config refers to registers and not bits.

I appreciate your help. Apparently, I misinterpreted "addr" as well. I thought that it would refer to Bits. That explains a lot :)
However, the "co size" is again in bits right?

When I now assume

janiversen, 2023-05-28:
Red the docs. And of course if something is not clear pull requests are welcome.

And no all addresses maps the physical memory and are therefore 16bit each.

janiversen, 2023-05-28:
I hope you are aware that the example server_simulator show how to use the simulator datastore, not the simulator with web etc. please read the doc.

janiversen, 2023-05-31:
Closing as it seems to be a non issue

closed 2023-05-31 by janiversen