pymodbus-dev/pymodbus #1553: Can't connect to a server using a usb dongle in Windows

pablomedina-frontera, 2023-05-28:
### Versions

- Python: 3.9.7
- OS: Windows 10
- Pymodbus: 3.2.2
- Modbus Hardware (if used): USB Dongle and weather station HongYuv

### Pymodbus Specific

- Client: ModbusSerialClient

### Description

I have a weather station connected to a USB Dongle. I was able to get values from that devices using modpoll:

```
modpoll -b 19200 -p none -a 10 -t   4:float -r 1 -c 20 COM3
```
```
modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2021 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: Modbus RTU, FC3
Slave configuration...: address = 10, start reference = 1, count = 20
Communication.........: COM3, 19200, 8, 1, none, t/o 1.00 s, poll rate 1000 ms
Data type.............: 32-bit float, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[1]: 0.000000
[3]: 0.000000
[5]: 24.060001
[7]: 38.923519
[9]: 945.440002
[11]: 0.000000
[13]: 0.000000
[15]: 0.000000
[17]: 0.000000
[19]: 0.000000
[21]: 0.000000
[23]: 0.000000
[25]: 0.000000
[27]: 0.000000
[29]: 0.000000
[31]: 0.000000
[33]: -26698129080320.000000
[35]: 0.000000
[37]: 0.000000
[39]: 0.000000
-- Polling slave... (Ctrl-C to stop)
```

However, I cannot get data from that device using Python, because I am receiving an exception:

</body>

</html>

```
Python 3.9.7 (tags/v3.9.7:1016ef3, Aug 30 2021, 20:19:38) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pymodbus.client import ModbusSerialClient
>>> import pymodbus
>>> pymodbus.pymodbus_apply_logging_config(level='debug', log_file_name='log.txt')
>>>
>>> serial = ModbusSerialClient(port='COM3',
...                             baudrate=19200,
...                             parity='0')
>>>
>>> serial.connect()
Traceback (most recent call last):
  File "J:\Seafile\Frontera\Desarrollos\Cólbun\codigos\venv\lib\site-packages\pymodbus\client\serial.py", line 254, in connect
    self.socket = serial.serial_for_url(
NameError: name 'serial' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "J:\Seafile\Frontera\Desarrollos\Cólbun\codigos\venv\lib\site-packages\pymodbus\client\serial.py", line 266, in connect
    except serial.SerialException as msg:
NameError: name 'serial' is not defined
```

My virtualenv has this:
```
(venv) J:\Seafile\Frontera\Desarrollos\Cólbun\codigos\backend>pip3 freeze
pymodbus==3.2.2
```

Let me know if you need more information. Thanks you guys.

Pablo

pablomedina-frontera, 2023-05-28:
I would like to comment I connected my dongle and the weather station to old laptop with Linux/Debian, and I got a "True" after serial.connect()

```
from pymodbus.client.sync import ModbusSerialClient
serial = ModbusSerialClient(port='/dev/ttyUSB0',
                            baudrate=19200,
                            parity='N')

serial.connect()
```

That  laptop has:
python 3.7.3
pymodbus 2.5.3
pyserial 3.5

Thanks

pablomedina-frontera, 2023-05-28:
I think I found the problem: after checking the virtualenv in my laptop with Debian, I have noticed pyserial was installed.
In my windows machine, pyserial was not installed after pymodbus installation. So I wrote:
```
pip3 install pyserial
```
and after that the code listed in my first message sent me a "True" after serial.connect()

Thanks,

Pablo

janiversen, 2023-05-28:
you have not installed pymodbus with the serial component, please read our README. you can trix it by doing "pip3 install serial"

pablomedina-frontera, 2023-05-28:
Hello:

I dropped my venv and I have created a new one. The installed libraries are as follow:

```
(venv) J:\Seafile\Frontera\Desarrollos\Cólbun\codigos\backend>pip3 freeze
future==0.18.3
iso8601==1.1.0
pymodbus==3.2.2
PyYAML==6.0
serial==0.0.97
```
If I ran the original code, I got the same problem (an exception)

If I dropped my venv once again, but this time pyserial is installed:
```
(venv) J:\Seafile\Frontera\Desarrollos\Cólbun\codigos\backend>pip3 freeze          
pymodbus==3.2.2
pyserial==3.5
```
Then I got the "True" after serial.connect()

Thanks

alexrudd2, 2023-05-30:
[serial](https://pypi.org/project/serial/) is a project to "A framework for serializing/deserializing JSON/YAML/XML into python class instances and vice versa", which is not what's needed here.

[pyserial](https://pypi.org/project/pyserial/) is a project for hardware serial lines.  Confusingly it installs a module named `serial`.

I have edited @janiversen's comment.  Also note you can install `pyserial` as an optional dependency of `pymodbus` itself by typing `pip3 install 'pymodbus[serial]'`

https://github.com/pymodbus-dev/pymodbus/blob/9c6acc385d2cb5362863cad740a90f693785f1e0/requirements.txt#L32-L33

closed 2023-05-30 by alexrudd2