pymodbus-dev/pymodbus #1561: Modbus simulator does not take into account min/max values limitations

t-dan, 2023-05-31:
### Versions

- Python: 3.11.3
- OS: ArchLinux
- Pymodbus: 3.2.2

### Pymodbus Specific

- Simulator

### Description

When using:
```
{
     "addr": [31, 32],
     "value": 50,
     "action": "random",
     "kwargs": {"min": 10, "max": 80},
}
```
or 
```
{
     "addr": [19, 20],
     "value": 14661,
     "action": "increment",
     "args": {"min": 1, "max": 100},
},
```
in a simulator configuration JSON (these copied from the definition of `default_config` in [test_simulator.py]( https://github.com/pymodbus-dev/pymodbus/tree/dev/test/test_simulator.py) ), it seems that the min/max values limitations are not taken into account.

I know that these are not in the official HTML documentation, but the limitations would be super-usefull.

janiversen, 2023-06-01:
???? It is not in the documentation because we have not implemented it, so no need to wonder why it does not work.

It could be a useful enhancements, so feel free to submit a PR.

janiversen, 2023-06-28:
On dev.

closed 2023-06-28 by janiversen