pymodbus-dev/pymodbus #1574: Possibly incorrect invocation of transport_connect in reconnect_timer

alexrudd2, 2023-06-02:
```
/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/events.py:73: 
RuntimeWarning: coroutine 'BaseTransport.transport_connect' was never awaited
```
I think this was most recently introduced in [2df9599](https://github.com/pymodbus-dev/pymodbus/pull/1541/files#diff-232ea64eea656684ba1d20bc100847b5fabad3d4627d4b8a51c49041ad84897dR374), but there's similar code earlier.
https://github.com/pymodbus-dev/pymodbus/blob/a68582168a863ca1d63cc0c7ecc3713d654cc9cd/pymodbus/transport/transport.py#L369-L372

This code calls `transport_connect` immediately, when I believe the intention is to pass it as an argument to `ensure_future` instead.  In other words, it should have no parentheses.

alexrudd2, 2023-06-02:
I don't know why the `pymodbus` tests don't catch this, but my library tests do.

janiversen, 2023-06-02:
I will have a look tomorrow, this could be the problem I have hunted a while.

alexrudd2, 2023-06-03:
Hmmm, I don't think my understanding is correct.

https://stackoverflow.com/a/48235721

janiversen, 2023-06-03:
I think you simply have misunderstood how async functions work.

When you call an async function immediately, that returns the future, but do not call the actual function.

then later you do future.call() to actually call the function and executes the code.

PR #1577 (2023-06-03): Proof transport reconnect works.

I back ported a part from #1572 in order to keep life simple.

Add new test file test_reconnect.py to verify reconnect.

closes #1574
closed 2023-06-03 by janiversen