pymodbus-dev/pymodbus #1578: AttributeError: 'AsyncModbusTcpClient' object has no attribute 'loop'

MrWaloo, 2023-06-05:
### Versions

- Python: 3.9.16 (under pyenv)
- OS: raspbian 10
- Pymodbus: 3.3.0

### Pymodbus Specific

- Client: tcp - async

### Description

When calling client.open(), an exception is raised.

### Code and Logs

```python
  async def connect(self):
    logging.debug('PyModbusClient: *' + self.eqConfig['name'] + '* connect called')
    try:
      if not self.connected or not self.client.connected:
        logging.debug('PyModbusClient: *' + self.eqConfig['name'] + '* connecting...')
        await self.client.connect()
      ret = True
    except Exception as e:
      logging.error('PyModbusClient: *' + self.eqConfig['name'] + '* Something went wrong while connecting to equipment id ' + self.eqConfig['id'] + ': ' + repr(e)) # + ' - ' + e.string)
      ret = False
    delay = float(self.eqConfig['eqFirstDelay'])
    await asyncio.sleep(delay)
    return ret
```

```
Traceback (most recent call last):
File "/var/www/html/plugins/mymodbus/ressources/mymodbusd/mymodbus.py", line 441, in connect
await self.client.connect()
File "/var/www/html/plugins/mymodbus/ressources/_pyenv/versions/3.9.16/lib/python3.9/site-packages/pymodbus/client/tcp.py", line 70, in connect
return await self.transport_connect()
File "/var/www/html/plugins/mymodbus/ressources/_pyenv/versions/3.9.16/lib/python3.9/site-packages/pymodbus/transport/transport.py", line 268, in transport_connect
self.call_connect_listen(),
File "/var/www/html/plugins/mymodbus/ressources/_pyenv/versions/3.9.16/lib/python3.9/site-packages/pymodbus/transport/transport.py", line 140, in 
self.call_connect_listen = lambda: self.loop.create_connection(
AttributeError: 'AsyncModbusTcpClient' object has no attribute 'loop'
```

janiversen, 2023-06-05:
I am thinking of a fix for v3.3.1, but I am not sure it is 100%.

MrWaloo, 2023-06-05:
The code is unchanged and worked with the last version. This has been reported by a user. I will upgrade my dev machine and test it if you want but the code hasn't been chaged since days and it worked for several users.

This function is the first to be called: asyncio is used:
```python
  def run(self, queue):
    self.queue = queue
    
    logging.getLogger('asyncio').setLevel(logging.WARNING)
    self.loop = asyncio.get_event_loop()
    
    if self.new_config is not None:
      self.apply_new_config()
    
    if self.eqConfig['eqRefreshMode'] == 'polling':
      self.loop.run_until_complete(self.run_polling())
      
    elif self.eqConfig['eqRefreshMode'] == 'cyclic':
      self.loop.run_until_complete(self.run_cyclic())
      
    elif self.eqConfig['eqRefreshMode'] == 'on_event':
      self.loop.run_until_complete(self.run_one())
      
    self.loop.close()
    self.shutdown()
```

Here is the shortest called function:
```python
  async def run_one(self):
    self.connected = False
    
    while not self.should_stop.is_set():
      # for time measuring
      t_begin = time.time()
      
      self.check_queue()
      await self.execute_write_requests(True, self.connected)
      
      if self.new_config is not None:
        self.apply_new_config()
      
      if self.read_cmd.is_set():
        self.read_cmd.clear()
        # Connect
        if not self.connected or not self.client.connected:
          self.connected = await self.connect()
        
        await self.read_all()
        
        # The loop has exited
        self.connected = await self.disconnect()
        
        self.cycle_times[self.cycle % 5] = time.time() - t_begin
```

janiversen, 2023-06-05:
With the fix you can instantiate without an async loop.

Clearly there are differences between v3.2.2 and v3.3.0 just look at the changelog, it includes a brand new transport layer for async clients. So it is natural that you need to do some adaptations in your program, if you do not want to do that then do not upgrade :-)

closed 2023-06-05 by janiversen

janiversen, 2023-06-05:
Your code "def run" is NOT async, you create a new loop with get_event_loop(), to test if you have a running loop use get_running_loop().

janiversen, 2023-06-05:
multiprocessing is not the same as asyncio.

PR #1579 (2023-06-05): Delay self.loop until connect().

fixes #1578