pymodbus-dev/pymodbus #1585: Running example give me an error

hairem, 2023-06-05:
### Versions

- Python:3.9
- OS:Raspbian
- Pymodbus:3.3.0
- Modbus Hardware (if used): Not Applicable at this point

### Pymodbus Specific

- Server: tcp-async

### Description
I have to rewrite a program that used an old version of pymodbus, so I am starting with an example that is similar to the original program, that being [Updating server example](https://pymodbus.readthedocs.io/en/latest/source/examples.html#updating-server) from your documentation. When I try to run an unedited version of this example I get this error: 
Traceback (most recent call last):
  File "/home/user/Flow/test.py", line 33, in <module>
    from examples.server_async import run_async_server, setup_server
ModuleNotFoundError: No module named 'examples'

Any idea what the issue is? I have already tried reinstalling the library.

janiversen, 2023-06-05:
Did you do we document in the top of examples download the examples directory, probably not because that is your problem.

hairem, 2023-06-05:
I did download the example folder and got the same error:
user@raspberrypi:~/pymodbus-dev/pymodbus-dev/examples $ ls
build_bcd_payload.py  client_sync.py       message_generator.py  server_simulator.py
certificates          client_sync_test.py  message_parser.py     server_sync.py
client_async.py       client_test.py       modbus_forwarder.py   server_updating.py
client_calls.py       contrib              server_async.py
client_custom_msg.py  helper.py            server_callback.py
client_payload.py     __init__.py          server_payload.py
user@raspberrypi:~/pymodbus-dev/pymodbus-dev/examples $ sudo python3 server_updating.py
Traceback (most recent call last):
  File "/home/user/pymodbus-dev/pymodbus-dev/examples/server_updating.py", line 33, in <module>
    from examples.server_async import run_async_server, setup_server
ModuleNotFoundError: No module named 'examples'

janiversen, 2023-06-05:
because you do not run the script inside the directory examples.

janiversen, 2023-06-06:
Closing the issue as it is not a pymodbus problem.

closed 2023-06-06 by janiversen

hairem, 2023-06-06:
I would suggest adding extra instructions for the pip installer because that's the installer I used and there were no additional instructions for downloading anything and running anything from a different folder.

janiversen, 2023-06-06:
The pip installer installs the library not the examples, just read the documentation

janiversen, 2023-06-06:
but pull request are welcome...however think about where examples get installed with pip, probably not where you want them.

hairem, 2023-06-06:
I've been using a previous version of your library for years now and I
never had to do all this to get the examples to work that is all I was
getting at. And I am reading your documentation but the fact that I have to
do the examples in the pi modbus Dev folder to get the examples to work
seems very counterintuitive when I'm trying to rewrite my program now to
meet this new library because I'm not going to be running my program in
your Dev folder.

janiversen, 2023-06-07:
Well it seems you have been using version 2.5 of the program, but that version also did not install the examples.

The difference is that in version 2.5.3 the examples was self contained and could just be copied. But as version 3.x is more complex we decided to go for a more normal examples set.

janiversen, 2023-06-07:
You will find that the shared part is
- helper.py which checks the command line
- (advanced clients) use client_async.py since that is where the client lib is explained.
- tls clients/servers need access to an example certificate 

so really for rewriting start with
- client_sync.py and remove get_comma line then you have what you need.

PR #1590 (2023-06-07): Simple examples.

fixes #1585