pymodbus-dev/pymodbus #1589: Modbus Error: [Input/Output] No Response

MGGEL, 2023-06-06:
### Versions

- Python: 3.7.3
- OS: Raspbian
- Pymodbus: 1.3.2
- Modbus Hardware (if used): Raspberrypi - Wago Pfc

### Pymodbus Specific

- Server: rtu - async
- Client: rtu - sync

### Description

I've tried to collect data from the server but i get allways the following message:
Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

### Logs

## Log

```
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0xa 0x1 0x3 0xe8 0x0 0x1 0x7c 0xc1
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xfb 0xfe 0xf 0xff 0x0
DEBUG:pymodbus.framer.rtu_framer:Frame check failed, ignoring!!
DEBUG:pymodbus.framer.rtu_framer:Resetting frame - Current Frame in buffer - 0xfb 0xfe 0xf 0xff 0x0
DEBUG:pymodbus.transaction:Getting transaction 10
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
```
## Code

```
import pymodbus
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.transaction import ModbusRtuFramer

import logging
import logging.handlers as Handlers

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient(
  method = 'rtu'
  ,port='/dev/ttyAMA0'
  ,stopbits = 1
  ,bytesize = 8
  ,baudrate=19200
  ,parity = 'E'
  ,timeout=1
  )
connection = client.connect()

coils = client.read_coils(1000, 1, unit=0xA)
print (coils)

register  = client.read_holding_registers(1001,1,unit=0xA)# start_address, count, slave_id
print (register)
```

janiversen, 2023-06-06:
unit= do not exist, did you actually read the documentation?

MGGEL, 2023-06-07:
I've read the documentation several times and foud unit= in some forum I think. 
I've tried several things, so there might be some parts which are acutally not exact like the documentation.

It doesn't make a difference if I remove the unit=.

```
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x3 0xe8 0x0 0x1 0x7c 0x6b
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xfb 0xfe 0xf 0x7b 0x0
DEBUG:pymodbus.framer.rtu_framer:Frame check failed, ignoring!!
DEBUG:pymodbus.framer.rtu_framer:Resetting frame - Current Frame in buffer - 0xfb 0xfe 0xf 0x7b 0x0
DEBUG:pymodbus.transaction:Getting transaction 0
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

```

janiversen, 2023-06-07:
Read the documentation of the library, not some forum! if you read forums make sure you they talk about the version you are using.

unit= have not existed for quite a while and have been replaced, if you just removed then that is probably your problem since your device seems to be running as space 10 (0xA).

janiversen, 2023-06-07:
If you read the documentation you will find both examples on how to use the library for this version as well as description of the parameters in each call.

Seems to me you have not read the documentation.

janiversen, 2023-06-07:
Closing this, since it is not a bug in the code (issues are for bugs, discussions are for the rest).

closed 2023-06-07 by janiversen