pymodbus-dev/pymodbus #1594: Connection closed after AsyncModbusSerialClient timeout

bsense-rius, 2023-06-12:
### Versions

- Python: 3.10
- OS: Ubuntu 22.04
- Pymodbus: 3.3.1
- Modbus Hardware (if used): GPIO Modbus

### Pymodbus Specific

- Client: rtu - async

### Description

When sending a read input registers and a timeout is triggered (device is not connected) the .connected property is False, even with close_comm_on_error=False

### Code and Logs
ModbusBaseClient._params(host=None, port='/dev/ttyUSB0', framer=<class 'pymodbus.framer.rtu_framer.ModbusRtuFramer'>, timeout=1.5, retries=3, retry_on_empty=False, close_comm_on_error=False, strict=True, broadcast_enable=False, kwargs={}, reconnect_delay=0, baudrate=9600, bytesize=8, parity='N', stopbits=1, handle_local_echo=False, source_address=None, sslctx=None, certfile=None, keyfile=None, password=None, server_hostname=None)

2023-06-12 02:25:50.345 DEBUG: Resetting frame - Current Frame in buffer -
2023-06-12 02:25:50.345 DEBUG: send: 0x1 0x4 0x0 0x1 0x0 0x2 0x20 0xb
2023-06-12 02:25:50.345 DEBUG: Adding transaction 1
2023-06-12 02:25:51.847 ERROR: TEMP] Connected? False

janiversen, 2023-06-12:
You cannot use ModbusBaseClient, please see documentation, you need to use the proper client for the desired transport e.g. ModbusSerialClient.

janiversen, 2023-06-12:
You are also referring to an interval variable (_params), so this is never going to work. Have a look at the example client_async.py

closed 2023-06-12 by janiversen

bsense-rius, 2023-06-12:
Thank you so much. In fact i use the AsyncModbusSerialClient as the title of this issue tells. Pretty much similar to what the example shows

I just posted the resulting parameters AFTER i executed the client.connect()

janiversen, 2023-06-12:
You were asked to poste the code, the parameters you refer to is used differently.

janiversen, 2023-06-12:
Please add the pymodbus debug log as pr documentation before you start using the library, and let's see what that says.