pymodbus-dev/pymodbus #1595: Modbus Forwarder Receiving but not Forwarding

SubToxic101, 2023-06-12:
### Versions

- Python: 3.9
- OS: Windows 10
- Pymodbus: 3.3.1
- Modbus Hardware: Two RS-485 Converters

### Pymodbus Specific

- Server: RTU - tried both sync/async
- Client: RTU - tried both sync/async

### Description

Relatively new to modbus and have been trying to implement a parity converter, that is a program that takes in Packets with say Even parity and sends out the same exact packets with Odd parity. End goal is to have the program spliced between an end device (address: 2) and the master to convert parities. Majority of the requests are going to be Write Multiple Registers requests.

To do this I tried setting up the PyModbus Forwarder example with the server configured for Even Parity, and the client configured for Odd parity. The Server receives all the traffic from my Master device, but there seems to be some issue on the client side. I am not receiving any packets on the end device and my RS-485 adapter TX/RX lights do not indicate any traffic. Switching both ends to even parity doesn't change anything, it just seems like the requests are getting stonewalled after being received by the server. How can I get the client to passthrough the masters messages that the server ahs received? It seems like the example may not be set up to forward a Write Multiple registers request? (Function Code 10). If not how would I set that up?

### Code and Logs

`import asyncio
import logging
import sys
from pymodbus.examples.helper import get_commandline
from pymodbus.client import AsyncModbusSerialClient
from pymodbus.datastore import ModbusServerContext
from pymodbus.datastore.remote import RemoteSlaveContext
from pymodbus.server import StartAsyncSerialServer

_logger = logging.getLogger()
_logger.setLevel("DEBUG")
_logger.addHandler(logging.StreamHandler(sys.stdout))

async def run_forwarder(client_port, port):
    """Run forwarder setup."""
    txt = f"### start forwarder, listen {port}, connect to {client_port}"
    _logger.info(txt)

    client = AsyncModbusSerialClient(
        port=client_port,
        baudrate=19200,
        parity="O"
    )
    await client.connect()
    assert client.connected
    print("done")
    # If required to communicate with a specified client use slave=<slave_id>
    # in RemoteSlaveContext
    # For e.g to forward the requests to slave with slave address 1 use
    # store = RemoteSlaveContext(client, slave=1)
    store = RemoteSlaveContext(client, slave=2)
    context = ModbusServerContext(slaves=store, single=True)
    print("done 2")
    await StartAsyncSerialServer(context=context, port=port, baudrate=19200, parity="E")
    # loop forever

if __name__ == "__main__":
    asyncio.run(run_forwarder(client_port="COM6", port="COM5")`

[image]

janiversen, 2023-06-12:
Are you sure your slaves are correct ?

remark at the moment we do not support write commands, but read should work.

SubToxic101, 2023-06-12:
Yes slaves are correct. If write is not supported that would explain it. Any workaround for passing through write commands?

janiversen, 2023-06-12:
If you need something urgent, you can try to fix it, however it is not trivial.

The other possibility is to start a server using the callback feature, and in there pass the call to the client...that too is not very simple.

The real solution, as I told another a couple of days ago, is to make a special framer, because at that level you have the request in the hand, and can pass it on the client, wait for the response (which you again will have as a byte stream) and return that.

If you fancy giving it a try, please make a pull request, so we can help out.

janiversen, 2023-06-12:
Thinking about it for 2 seconds more, if all you want is adapting the parity, you could use our new transport layer pymodbus/transport (see dev, the version released lacks functionality), with that you can easily interconnect 2 serial streams and make simple changes like parity.

SubToxic101, 2023-06-12:
Yes, currently all I need is the parity change. Could you give me some direction on how to utilize the transport layer? I don't fully understand how I would use the transport objects in my use case. If I open two serial connections how would I interconnect them?

janiversen, 2023-06-12:
Look at dev, Pymodbus/transport is the implementation 

pymodbus/client already use the transport layer

test/sub_transport should give you a pretty good idea on how to use it. 

Our upcoming forwarder needs to be able to convert between e.g. serial and tcp, and also support multiple clients, so that is more complicated

SubToxic101, 2023-06-13:
Apologies, still am a bit lost after looking it over. Do I have to instantiate a Null Modem and connect a client and server instance? Or do I have to replace the Transport inside of the Client and Server? Do I still keep the RemoteSlaveContexts / start the client and server as usual?

janiversen, 2023-06-13:
My suggestion was to only use transport, because you really do not care about the content of the message.

janiversen, 2023-06-13:
with the current null modem you cannot interconnect a modbus server and client, that will be available in a future PR.

I gave you 2 distint ideas (not solution), please don't start mixing them, that will not work.

janiversen, 2023-06-22:
I think this is solved by another contributor

janiversen, 2023-06-23:
I was a bit too fast, sorry.

janiversen, 2023-06-28:
Please try with newest dev.

janiversen, 2023-07-05:
Seems it is no longer a problem.

closed 2023-07-05 by janiversen