pymodbus-dev/pymodbus #1597: Pymodbus serial connection lost (synchronization?)

thesems, 2023-06-13:
### Versions

- Python: 3.11
- OS: Ubuntu 23.04
- Pymodbus: 3.3.1 (also tested on 3.2.1)
- Modbus Hardware (if used): Proprietary

### Pymodbus Specific

- Server: rtu (proprietary)
- Client: rtu - async (pymodbus)

### Description

We have 2 slave's with 2 unit id's.

On slave unit id 10, we have 2 periodic tasks running (a fast task with 0.5s poll rate and slow 3s poll rate).
On slave unit id 20, we have 1 periodic task running (a slow one with 5s polling rate).

I defined a single modbus client class to do modbus commands (read, write, ...).
Before calling pymodbus read/write, I use a asyncio.Lock to acquire/release for synchronization.
In this example I only used read_holding_registers. 

When running only unit ID 10, application runs OK.
When running only unit ID 20, applications runs OK.
When running both, after an arbitrary amount of time (depends on polling intervals), it will come to an error.

Please see code and logs below. With both active, I receive time-out often (time-out originates from the library).
My first guess would be that it is a synchronization error on my side, but after verifying my code and logs, it seems to be working fine.

Any help would be appreciated.

### Code and Logs

Log snippet below. Library logs contain "root:" label.

```
2023-06-13T14:04:21 DEBG  logging::debug:102] root: Getting transaction 42
[2023-06-13T14:04:21 TRACE client::read_holding_registers:149] mbHoldRead: [0, 2, 0, 19582, 0, 0, 0, 0, 0, 0, 0, 15000, 15000, 0, 0, 0, 0, 0, 0, 0, 0].
[2023-06-13T14:04:21 DEBG  client::read_holding_registers:117] Reading holding registers: 1915,2,33
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: send: 0x21 0x3 0x7 0x7b 0x0 0x2 0xb2 0x6
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: Adding transaction 33
[2023-06-13T14:04:21 TRACE mqtt_handler::send:170] Send on topic 1/time_since_boot: 19582.
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: recv: 0x21 0x3 0x4 0x0 0x0 0x1 0xac 0xda 0x1c
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: recv: 0x21 0x3 0x4 0x0 0x0 0x1 0xac 0xda 0x1c
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: Getting Frame - 0x3 0x4 0x0 0x0 0x1 0xac
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: Factory Response[ReadHoldingRegistersResponse': 3]
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: Frame advanced, resetting header!!
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: Getting transaction 33
[2023-06-13T14:04:21 TRACE client::read_holding_registers:149] mbHoldRead: [0, 428].
[2023-06-13T14:04:21 DEBG  client::read_holding_registers:117] Reading holding registers: 100,8,42
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: send: 0x2a 0x3 0x0 0x64 0x0 0x8 0x3 0xc8
[2023-06-13T14:04:21 DEBG  logging::debug:102] root: Adding transaction 42
[2023-06-13T14:04:22 ERROR client::read_holding_registers:130] Failed to read modbus 100 with count 8. Reason: time-out!
[2023-06-13T14:04:22 DEBG  logging::debug:102] root: Connection lost comm due to None
[2023-06-13T14:04:22 DEBG  logging::debug:102] root: Getting transaction 42
[2023-06-13T14:04:22 ERROR on_change_handler::on_missed_heartbeat:132] MCP heartbeat missed 3 times consecutively.
[2023-06-13T14:04:22 TRACE mqtt_handler::send:170] Send on topic 1/error_code: 50311.
[2023-06-13T14:04:22 TRACE mqtt_handler::send:170] Send on topic 1/online: 0.

```

A snippet from modbus_client.py:
```python

try:
    await self.lock.acquire()
    logger().debug(
        f"Reading holding registers: {address},{quantity},{unit}"
    )
    result = await self.client.read_holding_registers(  # type: ignore
        address=address, count=quantity, slave=unit
    )
except ModbusException as exc:
    logger().error(
        f"Failed to read modbus address {address} with count"
        f" {quantity}. Reason: modbus exception!"
    )
    raise exc
except asyncio.TimeoutError as exc:
    logger().error(
        f"Failed to read modbus {address} with count {quantity}."
        " Reason: time-out!"
    )
    raise exc
finally:
    self.lock.release()
```

janiversen, 2023-06-13:
try to use an asyncio loc.

thesems, 2023-06-14:
Hi, sorry, failed to mention that i use: self.lock = asyncio.Lock().

janiversen, 2023-06-14:
Well it is impossible to see from your supplied code, also it is impossible to see which Modbus class you use etc.

thesems, 2023-06-14:
I just managed to solve it, however I am not sure completely sure why it works.

I edited your example client_calls.py and made a loop which leads from the two devices.
The example below works, but remove the async.sleep, it will not work.

while True:
        await _handle_holding_registers_1(client)
        await asyncio.sleep(0.05)
        await _handle_holding_registers_2client)
        await asyncio.sleep(0.05)

It seems that the master/slave need a short break between the calls. When running asyncio tasks (create_task), even with asyncio.Lock, if it happens immediately after, it will cause an error. The fix for me also simple, as I just need to sleep for 0.05 before I release the lock.

janiversen, 2023-06-14:
yeah of course it needs a short break, that is what allows the task switching. Remember asyncio is preemptive task switching not threads.

thesems, 2023-06-14:
My assumption was that when the function _handle_holding_registers returns, it is by definition finished and ready to send another request. The sleep seems hackish a little bit. However, I am happy that it works now, so I would see this issue as closed from my perspective.

closed 2023-06-14 by janiversen