pymodbus-dev/pymodbus #1600: 0x11 report_slave_id got False and lost some bytes

mlango, 2023-06-14:
### Versions

- Python: 3.10
- OS: Win10
- Pymodbus: 3.3

### Pymodbus Specific

- Server: tcp

### Description

I want to send 0x11(17) to slave.

### Code and Logs

```python
client = ModbusTcpClient('192.168.0.1', port=502, timeout=3)
response = client.report_slave_id(1)

```
print(response)
```
ReportSlaveIdResponse(17, b'\x01\x02B\x00\x11\x01\x03', False)
```
Want does the False mean,and how can i decode it?
When i encode it i got
```
b'\x08\x01\x02B\x00\x11\x01\x03\x00'
```

I use modbus poll got this from my slave.
```
01 F9 00 00 00 0B 01 11 07 01 02 42 00 11 01 03 23 
```
It is a 23 at last.

mlango, 2023-06-14:
I use soket to seed it .I got a ‘#’ at last byte.Have you abandoned the #
```
b'\x00\x00\x00\x00\x00\x0b\x00\x11\x07\x01\x02B\x00\x11\x01\x03#'
```

janiversen, 2023-06-14:
Just controlled:
```
  01 F9 00 00 00 0B 01
```
is the tcp framer, the response begins with 0x11

what you got is the identifier.

You are aware the Report Slave ID is only defined for serial lines, as pr modbus documentation (we facilitate a link in our README)

janiversen, 2023-06-16:
all done ?

janiversen, 2023-06-17:
Closing as not relevant.

closed 2023-06-17 by janiversen