pymodbus-dev/pymodbus #1601: Async serial on windows times out

datid, 2023-06-14:
### Versions

- Python:3.11.3
- OS:Windows 10 AMD64
- Pymodbus:3.3.1
- Modbus Hardware (if used): OMEGA CN16DPT temperature controller https://assets.omega.com/manuals/M5451.pdf
-  https://assets.omega.com/manuals/M5458.pdf

### Pymodbus Specific

- Server: N/A - hardware PID controller
- Client: rtu - async - serial 485

### Description

Get timeout errors with async routines, which I don't see with equivalent sync commands.  Also the timeout disconnects the device.
Is it to do with the comments pymodbus/transport/serial_asyncio/__init__.py which suggest windows may not be supported. 

"""\
Support asyncio with serial ports.

Posix platforms only, Python 3.5+ only.

Windows event loops can not wait for serial ports with the current
implementation. It should be possible to get that working though.
"""

 I couldn't see anything in the docs that suggested this, but maybe I missed it.  Is there a work around ?

### Code and Logs

```python
# code and logs here.
import pymodbus
from pymodbus.client import AsyncModbusSerialClient,ModbusSerialClient
from pymodbus.transaction import ModbusRtuFramer 
import struct
import time

async def read_temp(client,slave=1):
    CURRENT_INPUT_VALUE=0x280
    t=time.time()
    try:
        response=await client.read_holding_registers(CURRENT_INPUT_VALUE,slave=slave,count=2)
        return struct.unpack('>f',struct.pack('>HH',*response.registers))[0],time.time()-t
    except TimeoutError:
        print(f"TIMEOUT!!! connected={client.connected}")
        return None,time.time()-t

async def test_modbus(port):
    pymodbus.pymodbus_apply_logging_config()
    client=AsyncModbusSerialClient(port,
                                   close_comm_on_error=False,
                                   framer=ModbusRtuFramer,
                                   baudrate=115200,
                                   parity='O',
                                   timeout=10,
                                   )
    await client.connect()
    t=0.0
    while client.connected:
        print(f"Temp is {await read_temp(client)}")

if __name__=="__main__":
    asyncio.run(test_modbus('COM4'))

# please use the following to format logs when posting them here
Temp is (24.886890411376953, 0.015622615814208984)
2023-06-14 11:35:32,826 DEBUG logging:102 send: 0x1 0x3 0x2 0x80 0x0 0x2 0xc4 0x5b
2023-06-14 11:35:32,826 DEBUG logging:102 Adding transaction 1
2023-06-14 11:35:32,842 DEBUG logging:102 recv: 0x1 0x3 0x4 0x41 0xc7 0x18 0x5a 0xd5 0xc9
2023-06-14 11:35:32,842 DEBUG logging:102 recv: 0x1 0x3 0x4 0x41 0xc7 0x18 0x5a 0xd5 0xc9
2023-06-14 11:35:32,842 DEBUG logging:102 Getting Frame - 0x3 0x4 0x41 0xc7 0x18 0x5a
2023-06-14 11:35:32,842 DEBUG logging:102 Factory Response[ReadHoldingRegistersResponse': 3]
2023-06-14 11:35:32,842 DEBUG logging:102 Frame advanced, resetting header!!
2023-06-14 11:35:32,842 DEBUG logging:102 Getting transaction 1
Temp is (24.886890411376953, 0.01562356948852539)
2023-06-14 11:35:32,842 DEBUG logging:102 send: 0x1 0x3 0x2 0x80 0x0 0x2 0xc4 0x5b
2023-06-14 11:35:32,842 DEBUG logging:102 Adding transaction 1
TIMEOUT!!! connected=False
Temp is (None, 9.982240438461304)
2023-06-14 11:35:42,934 DEBUG logging:102 Connection lost comm due to None
2023-06-14 11:35:42,934 DEBUG logging:102 Getting transaction 1
```

janiversen, 2023-06-14:
This might be correct, we just copied serial_async from the pyserial project, because they do not respond to pull requests.

If you think it is doable, please submit a pull request.

janiversen, 2023-06-14:
ok

closed 2023-06-14 by janiversen