pymodbus-dev/pymodbus #1605: Memory overflow occurs when AsyncModbusTcpClient loops to read holding registers

channelgao, 2023-06-15:
### Versions

- Python: 3.10
- OS: Windows 11
- Pymodbus: 3.3.1
- Modbus Hardware (if used): one RJ45 PLC Device

### Pymodbus Specific

- Client: tcp - async

### Description

I try to use asyncio to poll some holding registers in a PLC for value changes. But as the number of polls increases, the memory usage keeps getting bigger.

### Code and Logs

```python
# code and logs here.

import time
import asyncio

from pymodbus.client import AsyncModbusTcpClient

modbus_client = AsyncModbusTcpClient(host=TARGET_PLC_ADDRESS, port=502)

async def task(address, limit):
    connect_plc = AsyncModbusConnector(host)
    # connect PLC
    await connect_plc.connect()
    # read PLC data
    while True:
        new_address = address
        for _ in range(limit):
            new_address += 1
            # print(address)
            message = await modbus_client.read_holding_registers(slave=1, address=new_address, count=1)
            print(message)

async def run_async_client():
    start_address = 41008
    limit = 500
    # submit task
    task1 = asyncio.create_task(task(start_address, limit))
    await task1

if __name__ == '__main__':
    asyncio.run(run_async_client())
```

janiversen, 2023-06-15:
insert a short asyncio.sleep after the read, you are not giving your device a chance to respond.

janiversen, 2023-06-15:
You can also try to play with https://docs.python.org/3/library/gc.html

channelgao, 2023-06-15:
> ```python
> modbus_client
> ```

Thanks for your reply, I tried asyncio.sleep(), and del modbus_client  gc.collect(). But still same as usual

janiversen, 2023-06-15:
of course if you delete the garbage collection, the memory will grow. Anyhow as I wrote earlier it is really out of our hands all I can recommend is that you play with (not delete) garbage collection.

A slow running loop, should not accumulate a lot, but it still will accumulate, because the scope of your variables do not change.

How to you measure memory is growing ? because if virtual memory grows, your (in this case theoretical problem) is in your operating system, if it is committed memory then it is likely to be python (could still be the socket library not releasing buffers).

channelgao, 2023-06-16:
Thank you for your suggestion, I first deleted the modbus connection and then used gc.collect() to force garbage collection, and then started the modbus connection before the next loop is started. 

I mainly check the memory consumption through the windows resource manager. As you said, I now also think that the slow growth of memory usage caused by the program running in the same scope, because I did the same way but the polling time was enlarged to 5 seconds, which also caused the slow growth of memory usage. 

Next, I will try coroutine testing in different scopes. Thank you again!

janiversen, 2023-06-16:
Closing, as this is not a pymodbus issue, in theory it is a python issue on windows or a windows issue.

We have one very big modbus configuration that polls at a 100ms rate, this have been running on a small raspberry pi 4, for 3 months (together with Homeassistant), without any restarts or problems, like lack of memory

We have a relatively small modbus configuration that polls at a 100ms rate, this have been running on a different raspberry pi 4 together with Homeassistant for well over 9 months, without a single restart or problem.

closed 2023-06-16 by janiversen