pymodbus-dev/pymodbus #1606: AsyncModbusSerialClient is not working with multiple slaves concurrently

KarolyPoka, 2023-06-15:
### Versions

- Python: 3.9
- OS: Windows 10
- Pymodbus: 3.3.1
- Modbus Hardware: several RTU slave devices connected to a distribution box. The distribution box is connected to one USB serial serial port

### Pymodbus Specific
- Client: rtu- async

### Description

I am trying to read modbus registers from several slave devices concurrently, using asynchronous client. 

### Code and Logs
This version of the main() function is always executed with a timeout error
```python
import asyncio
import datetime
from pymodbus import pymodbus_apply_logging_config
from pymodbus.client import AsyncModbusSerialClient
from pymodbus.transaction import ModbusRtuFramer
from DeviceModbusRegisterMap import registers

async def run_async_client(port):
    """Run async client."""

    # activate debugging
    pymodbus_apply_logging_config("DEBUG")
    print("get client")

    client = AsyncModbusSerialClient(
        port,
        framer=ModbusRtuFramer,
        timeout=0.2,
        # retries=3,
        # retry_on_empty=False,
        # close_comm_on_error=False,
        # strict=True,
        baudrate=115200,
        # bytesize=8,
        # parity="N",
        stopbits=2,
        handle_local_echo=True,
    )

    print("connect to server")
    await client.connect()
    # test client is connected
    print("Client connected: ", client.connected)

    return client

async def run_calls(client, slave_address):
    debug_start = datetime.datetime.now()
    rr = await client.read_holding_registers(1288-1, 8, slave=slave_address)
    print(rr)
    debug_end = datetime.datetime.now()
    duration = debug_end - debug_start
    print("reading register time: ", duration.total_seconds())

async def run_trough(client, slave_address):
    debug_start = datetime.datetime.now()

    items = list(registers.items())
    slice = items[6:10]
    sliced_registers = dict(slice)

    for value in sliced_registers.values():
        rr = await client.read_holding_registers(value["adr"] - 1, value["len"], slave=slave_address)
[... 121 lines cut ...]
2023-06-15 13:49:50,321 DEBUG logging:102 Getting Frame - 0x3 0xc 0x0 0x2 0x0 0x0 0xc6 0x2a 0x43 0x96 0x0 0x0 0x0 0x0
2023-06-15 13:49:50,321 DEBUG logging:102 Factory Response[ReadHoldingRegistersResponse': 3]
2023-06-15 13:49:50,321 DEBUG logging:102 Frame advanced, resetting header!!
2023-06-15 13:49:50,321 DEBUG logging:102 Getting transaction 6
2023-06-15 13:49:50,322 DEBUG logging:102 send: 0x6 0x3 0x9 0xe7 0x0 0x6 0x77 0xd4
2023-06-15 13:49:50,323 DEBUG logging:102 Adding transaction 6
2023-06-15 13:49:50,336 DEBUG logging:102 recv: 0x6 0x3 0xc 0x0 0x0 0x0 0x80 0x66 0x11 0x43 0x55 0x0 0x0 0x0 0x0 0x30 0x48
2023-06-15 13:49:50,336 DEBUG logging:102 recv: 0x6 0x3 0xc 0x0 0x0 0x0 0x80 0x66 0x11 0x43 0x55 0x0 0x0 0x0 0x0 0x30 0x48
2023-06-15 13:49:50,336 DEBUG logging:102 Getting Frame - 0x3 0xc 0x0 0x0 0x0 0x80 0x66 0x11 0x43 0x55 0x0 0x0 0x0 0x0
2023-06-15 13:49:50,336 DEBUG logging:102 Factory Response[ReadHoldingRegistersResponse': 3]
2023-06-15 13:49:50,336 DEBUG logging:102 Frame advanced, resetting header!!
2023-06-15 13:49:50,336 DEBUG logging:102 Getting transaction 6
2023-06-15 13:49:50,336 DEBUG logging:102 send: 0x6 0x3 0xa 0x7 0x0 0x6 0x76 0x66
2023-06-15 13:49:50,337 DEBUG logging:102 Adding transaction 6
2023-06-15 13:49:50,351 DEBUG logging:102 recv: 0x6 0x3 0xc 0x0 0x2 0x0 0x0 0xc6 0xf 0x43 0x96 0x0 0x0 0x0 0x0 0xea 0x5e
2023-06-15 13:49:50,351 DEBUG logging:102 recv: 0x6 0x3 0xc 0x0 0x2 0x0 0x0 0xc6 0xf 0x43 0x96 0x0 0x0 0x0 0x0 0xea 0x5e
2023-06-15 13:49:50,351 DEBUG logging:102 Getting Frame - 0x3 0xc 0x0 0x2 0x0 0x0 0xc6 0xf 0x43 0x96 0x0 0x0 0x0 0x0
2023-06-15 13:49:50,351 DEBUG logging:102 Factory Response[ReadHoldingRegistersResponse': 3]
2023-06-15 13:49:50,351 DEBUG logging:102 Frame advanced, resetting header!!
2023-06-15 13:49:50,351 DEBUG logging:102 Getting transaction 6
6  reading register time:  0.061069

Process finished with exit code 0
```
I assume something goes wrong with the COM port handling when the second task starts, but I cant wrap my head around why the sleep solves this between the task creations??? How is it possible to read from several slaves asynchronously, connected to the same serial port? Thanks a lot in advance for your help!

janiversen, 2023-06-15:
You cannot run calls on one connection from 2 tasks, at least not unless you add some locking.

janiversen, 2023-06-15:
simply run the calls serialized, you only have one comm line so parallel will cause problem.

e.g. use one task, and have the 2 calls in there.

janiversen, 2023-06-16:
Close,  because it not a pymodbus issue but a wrong usage.

closed 2023-06-16 by janiversen

KarolyPoka, 2023-06-19:
Thanks a lot!