pymodbus-dev/pymodbus #1615: Changing Defaults.Slave before import pymodbus.client no longer works

gonzzor, 2023-06-18:
### Versions

- Python: 3.11.3
- OS: Fedora 37
- Pymodbus: 3.3.1
- Modbus Hardware (if used):

### Pymodbus Specific

- Client: rtu - sync

### Description

As of v3.1.0 it's no longer possible to set the default slave address using the `Defaults.Slave`.

This is due to commit 82f199de70f27fbce8620b6d452940b1de01d24b which hardcoded the default value of Slave to 0. Not sure why it was changed and what the purpose of Defaults.Slave is now?

### Code and Logs
The following code used to work fine with 3.0.2.
```import pymodbus.constants
pymodbus.constants.Defaults.Slave=1

import pymodbus.client
client=pymodbus.client.ModbusSerialClient("/dev/ttyACM0")
res = client.read_input_registers(address=0, count=1)
print(res.registers)
```

Found the commit by doing git bisect with the following script.
```#!/usr/bin/python
import sys
import inspect
def get_default_args(func):
    signature = inspect.signature(func)
    return {
        k: v.default
        for k, v in signature.parameters.items()
        if v.default is not inspect.Parameter.empty
    }

import pymodbus.constants
pymodbus.constants.Defaults.Slave=1

import pymodbus.client
client=pymodbus.client.ModbusSerialClient("/dev/ttyACM0")
v=(get_default_args(client.read_input_registers))
try:
    assert v['slave'] == 1, 'slave default parameter should be 1 but is %s' % (str(v['slave']))
except AssertionError as e:
    print(e)
    sys.exit(1)

print("All good!")``

janiversen, 2023-06-18:
Defaults.<x> are on their way out of the code.

And defaults was never intended to be modified externally, in many cases that will not work (due to the import sequence).

please use the slave= parameter in the calls.

closed 2023-06-18 by janiversen

janiversen, 2023-06-18:
Defaults.slave is still used in different parts of the code, but not the client.