pymodbus-dev/pymodbus #1629: ModbusSerialClient.close() function does not work ?

aurelienp69, 2023-06-26:
### Versions

- Python: 3.10.4
- OS: Windows 10
- Pymodbus: 3.2.2
- Modbus Hardware (if used): Custom device with embedded modbus server software in C.

### Pymodbus Specific

- Server: embedded (C)
- Client: rtu - sync

### Description
I'm a beginner in python software. I'm having an issue with the ModbusSerialClient.close() function that does not close the serial com port. Maybe I made a mistake but I reduce my test code to a very few lines, so I don't thing I did.

ModbusSerialClient.close() does not close the serial com port.
The COM port is not released.
The communication is still open, request are still answered.

### Code and Logs

```python
from pymodbus.client import ModbusSerialClient

client = ModbusSerialClient(method = "rtu", parity = "E", port = 'COM3', baudrate = 19200, timeout = 0.1, retries = 0, retry_on_empty = 0)
client.connect()
rr = client.diag_query_data(msg = [15], slave = 1)
print (rr.message[0])
client.close()
rr = client.diag_query_data(msg = [21], slave = 1)
print (rr.message[0])

### Output
PS C:\Users\Aurélien> & C:/Users/Aurélien/AppData/Local/Programs/Python/Python310/python.exe "x:/R&D/Autres App/Projets en Cours/A026 - SxIP/Logiciel/Pyhton Lib/close_test.py"
15
21

janiversen, 2023-06-26:
Sounds like you found a bug, which we will investigate (actually I thought it was in our test suite).

janiversen, 2023-06-26:
try to check er.is_error().

Furthermore please use our debug enabler (see documentation or template in the issue you submitted, That way you can see what is actually sent and received.

aurelienp69, 2023-06-27:
Hi janiversen, thanks for your reply.
At this point I don't understand what I shall do with rr.is_error()
But I succeeded in using the debug enabler. It shows nothing when executing the client.close() function.

### output:
PS C:\Users\Aurélien> & C:/Users/Aurélien/AppData/Local/Programs/Python/Python310/python.exe "x:/R&D/Autres App/Projets en Cours/A026 - SxIP/Logiciel/Pyhton Lib/close_test.py"
connected
2023-06-27 08:17:09,995 MainThread      DEBUG    logging        :102      Current transaction state - IDLE
2023-06-27 08:17:09,995 MainThread      DEBUG    logging        :102      Running transaction 1
2023-06-27 08:17:09,995 MainThread      DEBUG    logging        :102      SEND: 0x1 0x8 0x0 0x0 0x0 0xf 0xa0 0xf
2023-06-27 08:17:09,995 MainThread      DEBUG    logging        :102      New Transaction state "SENDING"
2023-06-27 08:17:09,996 MainThread      DEBUG    logging        :102      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-06-27 08:17:10,117 MainThread      DEBUG    logging        :102      Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-06-27 08:17:10,118 MainThread      DEBUG    logging        :102      RECV: 0x1 0x8 0x0 0x0 0x0 0xf 0xa0 0xf
2023-06-27 08:17:10,119 MainThread      DEBUG    logging        :102      Getting Frame - 0x8 0x0 0x0 0x0 0xf
2023-06-27 08:17:10,120 MainThread      DEBUG    logging        :102      Factory Response[DiagnosticStatusResponse': 8]
2023-06-27 08:17:10,121 MainThread      DEBUG    logging        :102      Frame advanced, resetting header!!
2023-06-27 08:17:10,121 MainThread      DEBUG    logging        :102      Adding transaction 1
2023-06-27 08:17:10,122 MainThread      DEBUG    logging        :102      Frame - [b'\x01\x08\x00\x00\x00\x0f\xa0\x0f'] not ready
2023-06-27 08:17:10,122 MainThread      DEBUG    logging        :102      Getting transaction 1
2023-06-27 08:17:10,123 MainThread      DEBUG    logging        :102      Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
15
before client.close
after client.close
connected
2023-06-27 08:17:10,253 MainThread      DEBUG    logging        :102      Current transaction state - TRANSACTION_COMPLETE
2023-06-27 08:17:10,254 MainThread      DEBUG    logging        :102      Running transaction 2
2023-06-27 08:17:10,256 MainThread      DEBUG    logging        :102      SEND: 0x1 0x8 0x0 0x0 0x0 0x15 0x21 0xc4
2023-06-27 08:17:10,259 MainThread      DEBUG    logging        :102      Changing state to IDLE - Last Frame End - None Current Time stamp - 1687846630.258062
2023-06-27 08:17:10,277 MainThread      DEBUG    logging        :102      New Transaction state "SENDING"
2023-06-27 08:17:10,282 MainThread      DEBUG    logging        :102      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-06-27 08:17:10,400 MainThread      DEBUG    logging        :102      Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-06-27 08:17:10,401 MainThread      DEBUG    logging        :102      RECV: 0x1 0x8 0x0 0x0 0x0 0x15 0x21 0xc4
2023-06-27 08:17:10,401 MainThread      DEBUG    logging        :102      Getting Frame - 0x8 0x0 0x0 0x0 0x15
2023-06-27 08:17:10,402 MainThread      DEBUG    logging        :102      Factory Response[DiagnosticStatusResponse': 8]
2023-06-27 08:17:10,403 MainThread      DEBUG    logging        :102      Frame advanced, resetting header!!
2023-06-27 08:17:10,404 MainThread      DEBUG    logging        :102      Adding transaction 1
2023-06-27 08:17:10,404 MainThread      DEBUG    logging        :102      Frame - [b'\x01\x08\x00\x00\x00\x15!\xc4'] not ready
2023-06-27 08:17:10,405 MainThread      DEBUG    logging        :102      Getting transaction 1
2023-06-27 08:17:10,405 MainThread      DEBUG    logging        :102      Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
21

janiversen, 2023-06-27:
look at our examples they explain among others, how to do a modbus call.

The log shows pymodbus is sending after close, and not just returning your parameter, so you have found a bug which we will investigate.

aurelienp69, 2023-06-27:
Ok thanks for that

janiversen, 2023-06-27:
Now I simulated your program, and the situation is a bit more complex. close() do close the serial port, but when you make a new modbus call it silently reconnects again.

This is a wanted behaviour, because in a running program, it is convenient that the library takes care of reconnecting, when you have a temporary outage.

closed 2023-06-27 by janiversen

aurelienp69, 2023-06-27:
From my point of view, the silent re-connection is an odd behavior.
But you're right I missed that the close function free the port. At the origin I had a problem with releasing the com port. But it should have been a mistake in my script. Then I start to test if the port was closed by launching requests what brought me to the conclusion that it doesn't release.
Thanks a lot for the explanation.

aurelienp69, 2023-06-27:
I have found some other issues in the library. This is not blocking because I have workarounds. Should I open issues about it ?

janiversen, 2023-06-27:
of course

janiversen, 2023-06-27:
You find it odd, others call it a feature :-) If you feel it should be different then you are welcome to submit a pull request, which we can review.

We are in the process of rewriting the async client (happens over several releases), and once that is done then the sync client will just be a call to the async client, allowing us to maintain less code.