pymodbus-dev/pymodbus #1631: ModbusUdpServer does not get loop from kwargs like other servers

znazter, 2023-06-28:
https://github.com/pymodbus-dev/pymodbus/blob/71ceb06305ef4ad8bbad582d8641dc268e0321cb/pymodbus/server/async_io.py#L528C43-L528C43

Please change to:

self.loop = kwargs.get("loop") or asyncio.get_event_loop()

janiversen, 2023-06-28:
Please submit a pull request if you want changes.

janiversen, 2023-06-28:
Be aware if you submit a Pull Request is should also update the documentation for the other server classes, because the loop= parameter is not documented in any of the classes.

The reason is simply that it is not needed if you run in an async environment, and in a sync environment the library handles it automatically.

closed 2023-06-28 by janiversen

znazter, 2023-06-28:
Hi again, 

Thank you for your reply, I am having trouble creating a branch, do i need any specific permissions to do so?