pymodbus-dev/pymodbus #1632: ModbusSerialClient.diag_read_slave_message_count do not take into account slave argument

aurelienp69, 2023-06-28:
### Versions

- Python: 3.10.4
- OS: Windows 10
- Pymodbus: 3.2.2
- Modbus Hardware (if used): Custom device with embedded modbus server software in C.

### Pymodbus Specific

- Server: embedded (C)
- Client: rtu - sync

### Description

diag_read_slave_message_count (and some other diag request) does not take the slave argument. The modbus address (fist byte) is always 0x00 (broadcast). Device doesn't answer to broadcast so I can't get the message count

### Code and Logs

```python
from pymodbus.client import ModbusSerialClient
import logging

FORMAT = ('%(asctime)-15s %(threadName)-15s' ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusSerialClient(method = "rtu", parity = "E", port = 'COM5', baudrate = 19200, timeout = 0.1, retries = 0, retry_on_empty = 0)
#client.connect()
#print ("connected")
rr = client.diag_read_slave_message_count(slave = 1)
print (vars (rr))
client.close()
```

### Output
2023-06-28 10:22:11,342 MainThread      DEBUG    logging        :102      Current transaction state - IDLE
2023-06-28 10:22:11,342 MainThread      DEBUG    logging        :102      Running transaction 1
2023-06-28 10:22:11,344 MainThread      DEBUG    logging        :102      SEND: 0x0 0x8 0x0 0xe 0x0 0x1 0x41 0xd9
2023-06-28 10:22:11,344 MainThread      DEBUG    logging        :102      New Transaction state "SENDING"
2023-06-28 10:22:11,345 MainThread      DEBUG    logging        :102      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-06-28 10:22:11,688 MainThread      DEBUG    logging        :102      Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)) 
2023-06-28 10:22:11,688 MainThread      DEBUG    logging        :102      Frame - [b''] not ready
2023-06-28 10:22:11,689 MainThread      DEBUG    logging        :102      Getting transaction 0
2023-06-28 10:22:11,689 MainThread      DEBUG    logging        :102      Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
{'fcode': 8, 'message': '[Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)', 'string': '[Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)'}

janiversen, 2023-06-28:
Did you read the modbus specification ? there are no slave argument in these requests.

janiversen, 2023-06-30:
"some other" is not particularly helpful, you need to write with which calls you have this problem.

closed 2023-06-30 by janiversen