pymodbus-dev/pymodbus #1633: ModbusSerialClient.diag_query_data does not support byte array argument for msg

aurelienp69, 2023-06-28:
### Versions
- Python: 3.10.4
- OS: Windows 10
- Pymodbus: 3.2.2
- Modbus Hardware (if used): Custom device with embedded modbus server software in C.

### Pymodbus Specific
- Server: embedded (C)
- Client: rtu - sync

### Description
ModbusSerialClient.diag_query_data only accept one integer for msg argument.
diag_query_data msg argument is specified to be a bytearray but it calls pdu_diag.ReturnQueryDataRequest that take an integer msg argument. According to Modbus specification, diag_query_data should takes a byte array (lenght limited by the maximum modbus frame lenght).

### Code and Logs
```python
from pymodbus.client import ModbusSerialClient

client.connect()
print ("connected")
rr = client.diag_query_data(msg = 15, slave = 1)
print (rr.message[0])
test_list = [2,3,4]
test_byte_array = bytearray (test_list)
rr = client.diag_query_data(msg = test_byte_array, slave = 1)
print (rr.message[0])
client.close()
```

### Output
Exception has occurred: error
required argument is not an integer
  File "X:\R&D\Autres App\Projets en Cours\A026 - SxIP\Logiciel\Pyhton Lib\query_test.py", line 18, in <module>
    rr = client.diag_query_data(msg = test_byte_array, slave = 1)
struct.error: required argument is not an integer

connected
2023-06-28 12:15:38,348 MainThread      DEBUG    logging        :102      Current transaction state - IDLE
2023-06-28 12:15:38,350 MainThread      DEBUG    logging        :102      Running transaction 1
2023-06-28 12:15:38,351 MainThread      DEBUG    logging        :102      SEND: 0x1 0x8 0x0 0x0 0x0 0xf 0xa0 0xf
2023-06-28 12:15:38,352 MainThread      DEBUG    logging        :102      New Transaction state "SENDING"
2023-06-28 12:15:38,355 MainThread      DEBUG    logging        :102      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-06-28 12:15:38,476 MainThread      DEBUG    logging        :102      Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-06-28 12:15:38,476 MainThread      DEBUG    logging        :102      RECV: 0x1 0x8 0x0 0x0 0x0 0xf 0xa0 0xf
2023-06-28 12:15:38,478 MainThread      DEBUG    logging        :102      Getting Frame - 0x8 0x0 0x0 0x0 0xf
2023-06-28 12:15:38,478 MainThread      DEBUG    logging        :102      Factory Response[DiagnosticStatusResponse': 8]
2023-06-28 12:15:38,480 MainThread      DEBUG    logging        :102      Frame advanced, resetting header!!
2023-06-28 12:15:38,481 MainThread      DEBUG    logging        :102      Adding transaction 1
2023-06-28 12:15:38,482 MainThread      DEBUG    logging        :102      Frame - [b'\x01\x08\x00\x00\x00\x0f\xa0\x0f'] not ready
2023-06-28 12:15:38,483 MainThread      DEBUG    logging        :102      Getting transaction 1
2023-06-28 12:15:38,483 MainThread      DEBUG    logging        :102      Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
15
2023-06-28 12:15:38,485 MainThread      DEBUG    logging        :102      Current transaction state - TRANSACTION_COMPLETE
2023-06-28 12:15:38,485 MainThread      DEBUG    logging        :102      Running transaction 2

janiversen, 2023-06-28:
Please submit a pull request if you want change, but please check with the modbus standard if e,g, bytearray is allowed.

aurelienp69, 2023-06-28:
Hi janiverse,
In the modbus standad (Modbus application protocol 1_1b3) it is said that the data is Nx2bytes.
I mention a byte array because the diag_query_data function documentation indicate that msg argument should be a byte array but byte arrays return an exception.

janiversen, 2023-06-28:
As I wrote feel free to submit a pull request alternatively wait until we come around to investigate it.

aurelienp69, 2023-06-28:
Ok! I'm new on Github and my python skills are poor. So I'm not sure to be able to pull a request that would be useful.

PR #1638 (2023-07-01): Restrict Return diag call to bytes.

Remark our server only allows 2 bytes.

fixes #1633
closed 2023-07-01 by janiversen