pymodbus-dev/pymodbus #1643: Async Modbus Timeout when using two slaves

cdsubanko, 2023-07-03:
### Versions

- Python: 3.11
- OS: Linux
- Pymodbus: 3.2.2
- Modbus Hardware (if used): N/A

### Pymodbus Specific

- Client: rtu - async

### Description
I am having a similar issue to #1597. 

I have two slaves (with the same setup and response time). 
Writing to them one after the other leads to a timeout. I need a sleep in between the write requests of at least 30ms to make it work. 
It has the same effect using await asyncio.sleep(0.03) or time.sleep(0.03). 

Although, it works if I write to the same slave twice, without any delay in between. 

My task is very time sensitive and therefore avoiding this 30ms delay would be good. 

In my opinion the "await" should wait unitl the response is received and only after this start the second write request. Is there anything that happens in the pymodbus library after returning the modbus response?

Code and logs can be seen below. 
Do you have any suggestions? 
Thank you in advance. 

### Code and Logs

My modbus client looks the following:
```python
class ModbusClient:
    def __init__(self, port: str):
        self.port = port
        self.loop = None
        self.client: AsyncModbusSerialClient | None = None
        self.error_cb = None
        self.lock = asyncio.Lock()

    async def setup(self, baudrate: int):
        self.client = AsyncModbusSerialClient(
            framer=ModbusRtuFramer,
            port=self.port,
            baudrate=baudrate,
            stopbits=1,
            parity="N",
            broadcast_enable=True,
        )
        await self.client.connect()

    def is_connected(self):
        return self.client is not None and self.client.connected

    async def read_holding_registers(self, address: int, quantity: int, unit: int, number_try: int = 0):
        result = await self.client.read_holding_registers(
            address=address, count=quantity, slave=unit
        )
        if result.isError():
            logger.error(f"Error while reading address {address}")
            ex = pymodbus.exceptions.ModbusException(result)
            raise ex
        return result`

    async def write(self, address: int, value: List[int], unit: int, number_try : int = 0):
        result = await self.client.write_registers(address=address, values=value, slave=unit)
        if unit != 0 and result.isError():  # Broadcast (unit 0) has no result
            logger.error(f"Error while writing address {address} value {value}")
            ex = pymodbus.exceptions.ModbusException(result)
            raise ex
        return result
```
I then try to run this simple test, where two modbus reads from two different slaves should be done:
```python
    @pytest_asyncio.fixture
    async def modbus(self) -> ModbusClient:
        modbus_client = ModbusClient(port="/dev/ttyUSB0")
        await modbus_client.setup(115200)
        return modbus_client

    @pytest.mark.asyncio
    async def test_multiple_reads(self, modbus:ModbusClient):
        result = await modbus.read_holding_registers(
            address=504, quantity=1, unit=1
        )
        result = await modbus.read_holding_registers(
            address=504, quantity=1, unit=2
        )
```

Which produces a TimeoutError. 

The logs are:
```
2023-06-30 10:30:09,802 DEBUG logging:102 send: 0x1 0x3 0x1 0xf8 0x0 0x1 0x4 0x7
2023-06-30 10:30:09,802 DEBUG logging:102 Adding transaction 1
2023-06-30 10:30:09,806 DEBUG logging:102 recv: 0x1 0x3 0x2 0x13 0x13 0xf4 0xb9
2023-06-30 10:30:09,806 DEBUG logging:102 Getting Frame - 0x3 0x2 0x13 0x13
2023-06-30 10:30:09,806 DEBUG logging:102 Factory Response[ReadHoldingRegistersResponse': 3]
2023-06-30 10:30:09,806 DEBUG logging:102 Frame advanced, resetting header!!
2023-06-30 10:30:09,806 DEBUG logging:102 Getting transaction 1
2023-06-30 10:30:09,806 DEBUG logging:102 Frame - [b'\x01\x03\x02\x13\x13\xf4\xb9'] not ready
2023-06-30 10:30:09,807 DEBUG logging:102 send: 0x2 0x3 0x1 0xf8 0x0 0x1 0x4 0x34
2023-06-30 10:30:09,807 DEBUG logging:102 Adding transaction 2
2023-06-30 10:30:12,810 INFO  logging:96 Serial lost connection.
2023-06-30 10:30:12,811 DEBUG logging:102 Client disconnected from modbus server: trying to send
2023-06-30 10:30:12,811 DEBUG logging:102 Getting transaction 2
2023-06-30 10:30:12,811 INFO  logging:96 Serial lost connection.
2023-06-30 10:30:12,811 WARNING logging:108 Ignoring launch of delayed reconnection, another is in progress
2023-06-30 10:30:12,812 DEBUG logging:102 Client disconnected from modbus server: None
2023-06-30 10:30:12,817 DEBUG logging:102 Waiting 100 ms before next connection attempt.
```

janiversen, 2023-07-03:
First of all you are not reading from 2 slaves, because unit= is no longer used please see the documentation.

Secondly if you read from real different slaves (devices) some devices needs a short pause to allow the switch.

janiversen, 2023-07-03:
It does seem that version 3.2.2 still converts unit= silently.

The problem is that the connection is lost for some reason, and adding a wait gives it time to reconnect.

Try to read multiple times from the slave once that works try to read correctly from multiple slaves.

cdsubanko, 2023-07-05:
Thank you for the quick response, the actual problem seems to lie in the modbus implementation of the two slaves/devices. The issue can be closed, thank you!

closed 2023-07-05 by cdsubanko