pymodbus-dev/pymodbus #1649: This library does not do what's advertised

enjoysmath, 2023-07-08:
Can't get this simple example to work:

```
#!/usr/bin/env python
"""
Pymodbus Server With Updating Thread
--------------------------------------------------------------------------

This is an example of having a background thread updating the
context while the server is operating. This can also be done with
a python thread::

    from threading import Thread

    thread = Thread(target=updating_writer, args=(context,))
    thread.start()
"""
# --------------------------------------------------------------------------- #
# import the modbus libraries we need
# --------------------------------------------------------------------------- #
#from pymodbus.server.asynchronous import StartSerialServer
from pymodbus.server import StartSerialServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer, ModbusAsciiFramer
import _thread

## --------------------------------------------------------------------------- #
## import the twisted libraries we need   sudo pip3 install Twisted
## --------------------------------------------------------------------------- #
#from twisted.internet.task import LoopingCall

# --------------------------------------------------------------------------- #
# configure the service logging
# --------------------------------------------------------------------------- #
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

# --------------------------------------------------------------------------- #
# define your callback process
# --------------------------------------------------------------------------- #

import RPi.GPIO as GPIO
import time

LED_PIN = 21

def updating_writer(a):
   """ A worker process that runs every so often and
   updates live values of the context. It should be noted
   that there is a race condition for the update.

   :param arguments: The input arguments to the call
   """
   log.debug("read coil")
   context = a[0]
   register = 1
   slave_id = 0x00
   address = 0x01
   values = context[slave_id].getValues(register, address, count=1)
   if values[0] == True:
      log.debug("LED ein")
      GPIO.output(LED_PIN, GPIO.HIGH)
   else:
      log.debug("LED aus")
      GPIO.output(LED_PIN, GPIO.LOW)

def run_updating_server():
   # ----------------------------------------------------------------------- # 
   # initialize your data store
   # ----------------------------------------------------------------------- # 

   store = ModbusSlaveContext(
       di=ModbusSequentialDataBlock(0, [17]*100),
        co=ModbusSequentialDataBlock(0, [17]*100),
        hr=ModbusSequentialDataBlock(0, [17]*100),
        ir=ModbusSequentialDataBlock(0, [17]*100))
   context = ModbusServerContext(slaves=store, single=True)

   # ----------------------------------------------------------------------- # 
   # initialize the server information
   # ----------------------------------------------------------------------- # 
   identity = ModbusDeviceIdentification()
   identity.VendorName = 'Pymodbus'
   identity.ProductCode = 'PM'
   identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
   identity.ProductName = 'pymodbus Server'
   identity.ModelName = 'pymodbus Server'
   identity.MajorMinorRevision = '2.2.0'

   # ----------------------------------------------------------------------- # 
   # run the server you want
   # ----------------------------------------------------------------------- # 
   time = 0.5  # 5 seconds delay
   loop = _thread.start_new_thread(target=updating_writer, args=(context,))
   #loop.start(time, now=False) # initially delay by time
   StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='/dev/ttyS0', timeout=2, baudrate=9600)

if __name__ == "__main__":
   GPIO.setmode(GPIO.BCM)
   GPIO.setwarnings(False)
   GPIO.setup(LED_PIN, GPIO.OUT)
   run_updating_server()
```

Well, from `pymodbus.server.sync` seems to deprecated so I tried:

closed 2023-07-08 by enjoysmath

janiversen, 2023-07-08:
Your title is quite provoking, so let me reply in the same manner:
- You do not do what you are supposed to do !

When you opened the issue, you were asked to supply a set of information (like version etc) in order for us to help you, you choose not to.

Neither did you bother to read our documentation, where we actually document how to use the library.

We provide a set of examples including an updating pymodbus server, which we guarantee work, because it is tested on every commit we make. These examples are all referenced in our documentation.

We do not guarantee, and never well, that every example ever written and present on the internet works, that is simply out of our hands.

The example you are using is based on a really old and unsupported version of pymodbus...and it actually still works with that version...but of course if you have installed a new version of pymodbus, you need to update the example.

The current version of the server is async, so using threads is not a very efficient way of doing updates.

And finally, every example on the internet is based on a specific version, so please use THAT version, instead of trying to stop a natural development of the library.

janiversen, 2023-07-08:
And I see you realised something and closed the issue yourself, thanks !