pymodbus-dev/pymodbus #1650: Bad performance with RTU serial from 3.2.0 onwards

mbs38, 2023-07-08:
### Versions

- Python: 3.9.2
- OS: Debian 11
- Pymodbus: dev Branch
- Modbus Hardware: Modbus RTU hardware I/O Module

### Pymodbus Specific

- Server: Modbus RTU hardware device
- Client: rtu - sync

### Description

After updating pymodbus I noticed that the performance in my application had deteriorated
significantly. Performing transactions takes more than twice the time it did before 3.2.0. I was able to achieve 28 transactions per second with 3.1.3. With 3.2.0 and it is only 11 transactions per second! For me this makes the library unusable.

### Code and Logs
I've written a minimal script do demonstrate the issue. 

## Script
import time
start = time.time()
from pymodbus.client import ModbusSerialClient
client = ModbusSerialClient(port="/dev/ttyUSB0",baudrate=38400,parity="N")
for x in range(0,100):
    rr=client.read_holding_registers(0,1,224)
print(time.time()-start)

## Results
git checkout dev
python3 modbustest.py
8.997401237487793

git checkout v3.2.0
python3 modbustest.py
8.967790842056274

git checkout v3.1.3
python3 modbustest.py
3.569199323654175

## Further reading
And just for comparison when using modpoll by modbusdriver.com:

import time
import os

start = time.time()
for x in range(0,100):
    os.system('./modpoll -m rtu -p none -b 38400 -a 224 -1 /dev/ttyUSB0')
print(time.time()-start)

When I run this I get: 1.6396584510803223 
This means this tool achieves 61 transactions per second. Maybe pymodbus should be aiming for that kind of performance :-)

janiversen, 2023-07-08:
That is strange because there were no changes relating to the transfer in version 3.2. Between 3.2 and 3.3.2 we did correct a couple of bugs, but as far as I know nothing performance related.

janiversen, 2023-07-08:
Your problem could be that you do not specify slave= in the call.

Please make debug log, as you were asked to, when you created the issue. Without a debug log we have no way of helping you.

I have a serial connection running with 155.200 bps on a rpi 3. I read about 10 registers every second, 50 registers every 2 seconds and do file read (load profiles) every minute without problems.

mbs38, 2023-07-08:
Thanks for the quick response! I will try to provide some more info.

janiversen, 2023-07-08:
Your example suggest you do not change the test program between the versions, that is surely causing some problems. We main API_changes.txt to show what should be changed between versions.

mbs38, 2023-07-08:
> Your example suggest you do not change test program between the versions
Exactly. But I don't see any changes in [API_changes.rst](https://github.com/pymodbus-dev/pymodbus/blob/dev/API_changes.rst) that might apply to this example. What would you have me change between v3.1.3 and v3.2.0?

Changing  rr=client.read_holding_registers(2,1,224) to rr=client.read_holding_registers(2,1,slave=224) does not change anything.

[modbustest.log](https://github.com/pymodbus-dev/pymodbus/files/11991289/modbustest.log)

janiversen, 2023-07-08:
No you are right, apart from slave= there are no changes, simply because we have not touched the sync client, we are working heavily on the async client. So I am looking for any reason why you experience that the client is slower. The changes I remember is on dev.

You have not added debug as pr documentation, so I cannot see the timing of the calls, but I also cannot see any errors in the log.

I have no idea why your system have so low a performance (since there are no code changes), I can only suggest you stay at 3.1.3...it does not seem you have a need for the new things in 3.2, 3.3 and 3.4 (dev).

Once we have refactored the async client, it will be integrated into the sync client (the sync client will just be a sync cover class), and at least right now the new async client is double as fast as version 3.3.2

We do aim for performance but not over stability, having said that pull requests are more than welcome.

mbs38, 2023-07-15:
Hi, 
I am willing to provide any information you need. 

> You have not added debug as pr documentation, so I cannot see the timing of the calls, but I also cannot see any errors in the log.

What exactly would you like me to provide and how?

janiversen, 2023-07-16:
I already wrote it, and when you opened the issue the template asked you to do it.

mbs38, 2023-07-16:
You're right. An oversight on my part. Sorry.

Script:

```
import time
start = time.time()

import pymodbus
pymodbus.pymodbus_apply_logging_config()
from pymodbus.client import ModbusSerialClient

client = ModbusSerialClient(port="/dev/ttyUSB0",baudrate=38400,parity="N")

for x in range(0,100):
    rr=client.read_holding_registers(2,1,slave=224)
print(time.time()-start)
```

Logs:

[dev.log](https://github.com/pymodbus-dev/pymodbus/files/12063832/dev.log)

janiversen, 2023-07-16:
Nothing unusual in the log, but a couple of observations:
- the time from pymodbus send to receive is average 86 msec, that is quite a long time spent in OS, usb and device.
- the between app notification and next send is cerca 10 msec.
This means the turnaround time is around 100 msec.

You compare with modpoll, which seems highly unfair. As far as I know modpoll is programmed in C which is a factor faster than Python. Modpoll do not have an api, meaning the save lots of code, by operating directly on the message structures.

You also compare 3.1.3 and 3.2.0, which is more interesting, especially since I am unaware of bigger changes in the sync client, but feel free to diff the 2 versions (you should look for client/base.py and client/serial.py, but only the sync part).

I have marked this as a bug.

closed 2023-07-27 by janiversen