pymodbus-dev/pymodbus #1653: I can not use Pymodbus , ModbusSerial to Connect PLC

Bestpj12, 2023-07-12:
### Versions

- Python: 3.11.4
- OS: Windows
- Pymodbus:PIP [ Visual Studio Code ]
- Modbus Hardware (if used): PLC HNC

### Pymodbus Specific

- Client: rtu 

### Description

I want to Connect My Python to My PLC via Pymodbus ModbusSerial But It seem not working Please Help me But When I use Pymodbus , ModbusTcp It working Can read and write data to plc correctly 

### Code and Logs

```python
from pymodbus.client import ModbusSerialClient  
client=ModbusSerialClient(port='COM5',framer='rtu',baudrate=9600,bytesize=8,parity='N',stopbits=2) 

client.connect()
result=client.read_holding_registers(address=512,count=1,slave=0x02)

client.close()

# Output
[![output.jpeg](https://cdn.pic.in.th/file/picinth/output.jpeg)](https://pic.in.th/image/LnjLq)

![output](https://github.com/pymodbus-dev/pymodbus/assets/139318553/ae53ab7c-9c54-4d7a-8657-b98dd7913ad1)

janiversen, 2023-07-12:
Plase tell us which version you are using, and add debug log pr documentation

janiversen, 2023-07-12:
We dot have access to the output you posted, and an image do not help.

Bestpj12, 2023-07-12:
1.pymodbus version 3.3.2 

2.the output I insert  is the Picture link of Terminal when I Run Script Python On Visual Studio Code  I can not Copy Terminal on Visual Studio Code

janiversen, 2023-07-12:
Run the pymodbus debugger as requested.

Bestpj12, 2023-07-13:
You mean the error code that show on terminal when I run script python right 
1.	__init__ (framer,**kwargs)
2.	self.framer=self.params.framer(ClientDecoder(),self)

Could you possibly code it for me to see? I'm very new to python and confused the detail of Modbus serial

janiversen, 2023-07-13:
No I mean read the documentation, when you opened this issue, the template text told you exactly how to do it.

pymodbus_apply……..

janiversen, 2023-07-19:
Is this still open ?

closed 2023-07-19 by janiversen