pymodbus-dev/pymodbus #1658: minval and maxval for float32 in simulator

RyIot3329, 2023-07-14:
### Versions

- Python: Python 3.9.2
- OS: Debian 11
- Pymodbus: dev

### Pymodbus Specific

- Server: tcp - Simulator

### Description

I have this defined in my json file for the simulator.

        {
          "addr": [7, 8],
          "value": 72.0,
          "action": "random",
          "kwargs": { "minval": 65.0, "maxval": 78.0 }
        }

It does not seem to follow the minval and maxval values from what I can see. I am getting random Float values, so that works. But it isn't in between the bounds. Any ideas?

janiversen, 2023-07-15:
Which version are you using ? that might be the problem.

janiversen, 2023-07-15:
datastore/simulator.py#633 it seems to do what it is supposed to do.

I need to check it.

janiversen, 2023-07-15:
I see it's not covered in our test suite, I will add a specific test and see what happens.

janiversen, 2023-07-15:
I have now added a test case, and as far as I can see everything works as it should.

Did you make the configuration within the "float32" label ?

RyIot3329, 2023-07-15:
Yeah I added it within the "float" block. Would it help to upload the entire json file?

janiversen, 2023-07-15:
No, a debug log would help.

RyIot3329, 2023-07-17:
Sorry for the delay on this. Here is the log file and the JSON file I am using
[log.txt](https://github.com/pymodbus-dev/pymodbus/files/12072540/log.txt)
[tstat.txt](https://github.com/pymodbus-dev/pymodbus/files/12072544/tstat.txt)

PR #1659 (2023-07-15): test action_increment.

fixes #1658
PR #1671 (2023-07-19): Add new example simulator server/client.

fixes #1658
closed 2023-07-19 by janiversen