pymodbus-dev/pymodbus #1666: [Enhancements] Logging function only output to file (when filename is provided)

DivineMK, 2023-07-19:
### Versions

- Python: 3.11.4
- OS: Windows
- Pymodbus: 3.3.2
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp - sync/async
- Client: tcp - sync/async

### Description

Need logger.basicConfig() for logs to appear in console.
### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config()
```

janiversen, 2023-07-19:
Pull requests are welcome.

The current implementation sends everything to the console and that works as intended. You can always redirect the output using e.g. tee dump.log, and get it on file as well.

DivineMK, 2023-07-19:
When I ran the original example, it only has 1 line in the console: 2023-07-19 10:11:56,519 INFO  logging:96 Server(TCP) listening. The two lines are missing:
2023-07-19 10:04:33,582 INFO  server_async:76 ### Create datastore
2023-07-19 10:04:33,595 INFO  server_async:157 ### start ASYNC SERVER, listening on :5020 - tcp

janiversen, 2023-07-19:
Yes because debug is not enabled pr default. you need to use '--log debug' to activate the debug info.

DivineMK, 2023-07-19:
I tried again with MSYS2 Python 3.10.11 and add --log debug there is still no info.
My code:
```python
#!/usr/bin/env python3
"""Pymodbus asynchronous Server Example.

An example of a multi threaded asynchronous server.

usage: server_async.py [-h] [--comm {tcp,udp,serial,tls}]
                       [--framer {ascii,binary,rtu,socket,tls}]
                       [--log {critical,error,warning,info,debug}]
                       [--port PORT] [--store {sequential,sparse,factory,none}]
                       [--slaves SLAVES]

Command line options for examples

options:
  -h, --help            show this help message and exit
  --comm {tcp,udp,serial,tls}
                        "serial", "tcp", "udp" or "tls"
  --framer {ascii,binary,rtu,socket,tls}
                        "ascii", "binary", "rtu", "socket" or "tls"
  --log {critical,error,warning,info,debug}
                        "critical", "error", "warning", "info" or "debug"
  --port PORT           the port to use
  --baudrate BAUDRATE   the baud rate to use for the serial device
  --store {sequential,sparse,factory,none}
                        "sequential", "sparse", "factory" or "none"
  --slaves SLAVES       number of slaves to respond to

The corresponding client can be started as:
    python3 client_sync.py
"""
import asyncio
import logging

from examples import helper
from pymodbus import __version__ as pymodbus_version
from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusServerContext,
    ModbusSlaveContext,
    ModbusSparseDataBlock,
)
from pymodbus.device import ModbusDeviceIdentification

# --------------------------------------------------------------------------- #
# import the various client implementations
# --------------------------------------------------------------------------- #
from pymodbus.server import (
    StartAsyncSerialServer,
    StartAsyncTcpServer,
    StartAsyncTlsServer,
    StartAsyncUdpServer,
)

from pymodbus import pymodbus_apply_logging_config

# logging.basicConfig()
_logger = logging.getLogger(__name__)
pymodbus_apply_logging_config(level=logging.DEBUG, log_file_name="pymodbus_async_server.log")
# log_formatter = logging.Formatter("%(asctime)s %(levelname)-5s %(module)s:%(lineno)s %(message)s")

# log_stream_handler = logging.StreamHandler()
# log_stream_handler.setFormatter(log_formatter)
# _logger.addHandler(log_stream_handler)

# log_file_handler = logging.FileHandler(mode="a", filename="pymodbus_async_server.log")
# log_file_handler.setFormatter(log_formatter)
# _logger.addHandler(log_file_handler)
[... 143 lines cut ...]
            # port=port,  # on which port
            identity=args.identity,  # server identify
            # custom_functions=[],  # allow custom handling
            address=address,  # listen address
            framer=args.framer,  # The framer strategy to use
            certfile=helper.get_certificate(
                "crt"
            ),  # The cert file path for TLS (used if sslctx is None)
            # sslctx=sslctx,  # The SSLContext to use for TLS (default None and auto create)
            keyfile=helper.get_certificate(
                "key"
            ),  # The key file path for TLS (used if sslctx is None)
            # password="none",  # The password for for decrypting the private key file
            # ignore_missing_slaves=True,  # ignore request to a missing slave
            # broadcast_enable=False,  # treat slave_id 0 as broadcast address,
            # timeout=1,  # waiting time for request to complete
            # TBD strict=True,  # use strict timing, t1.5 for Modbus RTU
        )
    return server

if __name__ == "__main__":
    run_args = setup_server(description="Run asynchronous server.")
    asyncio.run(run_async_server(run_args), debug=True)
```
I also set the level to the debug before so I don't think missing --log debug is the issue.

janiversen, 2023-07-19:
hmmm I stand corrected seems you found a bug.

janiversen, 2023-07-19:
The examples do not use our internal debugging, but a normal logger:
```
_logger = logging.getLogger()
_logger.setLevel("DEBUG")
```
Which should not be affected by how we log internally.

janiversen, 2023-07-19:
Solved in https://github.com/pymodbus-dev/pymodbus/commit/9e23eb95ce8c9b622b538b1491129b87c07ff5f5

closed 2023-07-19 by janiversen