pymodbus-dev/pymodbus #1668: [Bug] Client calls test diagnostic request failed

DivineMK, 2023-07-19:
### Versions

- Python: 3.11.2
- OS: Windows
- Pymodbus: 3.32

### Pymodbus Specific

- Server: tcp - sync/async
- Client: tcp - sync/async

### Description

I ran server_async.py and then client_calls.py (I only added logging function) but it has an error (see below).
### Code and Logs
client_calls.py:

```python
#!/usr/bin/env python3
"""Pymodbus Client modbus call examples.

Please see:

    async_template_call

    template_call

for a template on how to make modbus calls and check for different
error conditions.

The _handle_.... functions each handle a set of modbus calls with the
same register type (e.g. coils).

All available modbus calls are present. The difference between async
and sync is a single 'await' so the calls are not repeated.

If you are performing a request that is not available in the client
mixin, you have to perform the request like this instead:

from pymodbus.diag_message import ClearCountersRequest
from pymodbus.diag_message import ClearCountersResponse

request  = ClearCountersRequest()
response = client.execute(request)
if isinstance(response, ClearCountersResponse):
    ... do something with the response

This example uses client_async.py and client_sync.py to handle connection,
and have the same options.

The corresponding server must be started before e.g. as:

    ./server_async.py
"""
import asyncio
import logging

import pymodbus.diag_message as req_diag
import pymodbus.mei_message as req_mei
import pymodbus.other_message as req_other
from examples.client_async import run_async_client, setup_async_client
from examples.client_sync import run_sync_client, setup_sync_client
from pymodbus.exceptions import ModbusException
from pymodbus.pdu import ExceptionResponse

# _logger = logging.getLogger()
# _logger.setLevel("DEBUG")

_logger = logging.getLogger(__name__)
log_formatter = logging.Formatter("%(asctime)s %(levelname)-5s %(module)s:%(lineno)s %(message)s")
[... 507 lines cut ...]
  File "D:\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\Documents\Programming\Modbus\pymodbus\examples\client_calls.py", line 311, in async_helper
    await run_async_client(testclient, modbus_calls=run_async_calls)
  File "D:\Documents\Programming\Modbus\pymodbus\examples\examples\client_async.py", line 125, in run_async_client
    await modbus_calls(client)
  File "D:\Documents\Programming\Modbus\pymodbus\examples\client_calls.py", line 300, in run_async_calls
    await _execute_diagnostic_requests(client)
  File "D:\Documents\Programming\Modbus\pymodbus\examples\client_calls.py", line 239, in _execute_diagnostic_requests
    await client.execute(
  File "D:\Python\Python311\Lib\site-packages\pymodbus\client\base.py", line 187, in async_execute
    packet = self.framer.buildPacket(request)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Python\Python311\Lib\site-packages\pymodbus\framer\socket_framer.py", line 211, in buildPacket
    data = message.encode()
           ^^^^^^^^^^^^^^^^
  File "D:\Python\Python311\Lib\site-packages\pymodbus\diag_message.py", line 92, in encode
    packet += struct.pack(">H", piece)
              ^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: required argument is not an integer
```

janiversen, 2023-07-19:
We are not yet supporting python 3.11, so it might be a problem due to a 3.11 change.

We do test client_calls.py with 3.8, 3.9 and 3.10 where it works as it should.

Please test with python 3.10 and check if you still have a problem.

If you want it to run on python 3.11, then pull requests are welcome. The reason we currently do not support 3.11 is that github CI do not support 3.11

DivineMK, 2023-07-19:
I tried again with MSYS2 Python 3.10.11 but I got the same error

janiversen, 2023-07-19:
Then you have a different problem like e.g. the requirements is not installed. Or CI test the functionality with every commit, so an error like you have should not pass

janiversen, 2023-07-19:
I assume you are aware that diag does not work on serial.

janiversen, 2023-07-19:
Try with dev, this is server_async.py:
```
(pymodbus) jan@MacMini examples % ./server_async.py            
INFO:/Users/jan/repos/pymodbus/examples/./server_async.py:Starting...
INFO:/Users/jan/repos/pymodbus/examples/./server_async.py:### Create datastore
INFO:/Users/jan/repos/pymodbus/examples/./server_async.py:### start ASYNC server, listening on 5020 - tcp
2023-07-19 18:26:06,483 INFO  logging:96 Server(TCP) listening.
INFO:pymodbus.logging:Server(TCP) listening.
```

and this is client_calls.py:
```
(pymodbus) jan@MacMini examples % ./client_calls.py 
INFO:/Users/jan/repos/pymodbus/examples/client_async.py:### Create client object
INFO:/Users/jan/repos/pymodbus/examples/client_async.py:### Client starting
DEBUG:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Template coils response: [True, False, False, False, False, False, False, False]
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Reading Coil different number of bits (return 8 bits multiples)
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Write false/true to coils and read to verify
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Write False to address 1-8 coils
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Reading discrete input, Read address:0-7
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### write holding register and read holding registers
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### write read holding registers, using **kwargs
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### read input registers
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Running information requests.
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Running diagnostic requests.
INFO:/Users/jan/repos/pymodbus/examples/client_async.py:### End of Program
INFO:/Users/jan/repos/pymodbus/examples/client_sync.py:### Create client object
INFO:/Users/jan/repos/pymodbus/examples/client_sync.py:### Client starting
DEBUG:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Template coils response: [True, False, False, False, False, False, False, False]
INFO:/Users/jan/repos/pymodbus/examples/client_sync.py:### End of Program
```

And with v3.3.2 it also works. Server_async.py output:
```
(pymodbus) jan@MacMini examples % ./server_async.py 
2023-07-19 18:28:19,201 INFO  logging:96 Server(TCP) listening.
```

client_calls.py output:
```
(pymodbus) jan@MacMini examples % ./client_calls.py 
INFO:/Users/jan/repos/pymodbus/examples/client_async.py:### Create client object
INFO:/Users/jan/repos/pymodbus/examples/client_async.py:### Client starting
DEBUG:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Template coils response: [True, False, False, False, False, False, False, False]
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Reading Coil different number of bits (return 8 bits multiples)
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Write false/true to coils and read to verify
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Write False to address 1-8 coils
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Reading discrete input, Read address:0-7
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### write holding register and read holding registers
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### write read holding registers, using **kwargs
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### read input registers
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Running information requests.
INFO:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Running diagnostic requests.
INFO:/Users/jan/repos/pymodbus/examples/client_async.py:### End of Program
INFO:/Users/jan/repos/pymodbus/examples/client_sync.py:### Create client object
INFO:/Users/jan/repos/pymodbus/examples/client_sync.py:### Client starting
DEBUG:/Users/jan/repos/pymodbus/examples/./client_calls.py:### Template coils response: [True, False, False, False, False, False, False, False]
INFO:/Users/jan/repos/pymodbus/examples/client_sync.py:### End of Program
(pymodbus) jan@MacMini examples % 
(pymodbus) jan@MacMini examples % 
(pymodbus) jan@MacMini examples % 
(pymodbus) jan@MacMini examples % ./client_calls.py 
(pymodbus) jan@MacMini examples % ./client_calls.py --log debug
2023-07-19 18:28:32,629 DEBUG logging:102 Connecting to 127.0.0.1:5020.
2023-07-19 18:28:32,629 DEBUG logging:102 Connecting comm
2023-07-19 18:28:32,630 DEBUG logging:102 Connected to comm
2023-07-19 18:28:32,630 DEBUG logging:102 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x1 0x0 0x1
2023-07-19 18:28:32,630 DEBUG logging:102 Adding transaction 1
2023-07-19 18:28:32,631 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x1
2023-07-19 18:28:32,631 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x1
[... 265 lines cut ...]
2023-07-19 18:28:32,645 DEBUG logging:102 Current transaction state - IDLE
2023-07-19 18:28:32,645 DEBUG logging:102 Current transaction state - IDLE
2023-07-19 18:28:32,645 DEBUG logging:102 Running transaction 1
2023-07-19 18:28:32,645 DEBUG logging:102 Running transaction 1
2023-07-19 18:28:32,645 DEBUG logging:102 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x1 0x0 0x1
2023-07-19 18:28:32,645 DEBUG logging:102 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x1 0x0 0x1
2023-07-19 18:28:32,645 DEBUG logging:102 New Transaction state "SENDING"
[... cut ...]
2023-07-19 18:28:32,646 DEBUG logging:102 Factory Response[ReadCoilsResponse': 1]
2023-07-19 18:28:32,646 DEBUG logging:102 Adding transaction 1
2023-07-19 18:28:32,646 DEBUG logging:102 Adding transaction 1
2023-07-19 18:28:32,646 DEBUG logging:102 Getting transaction 1
2023-07-19 18:28:32,646 DEBUG logging:102 Getting transaction 1
2023-07-19 18:28:32,646 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2023-07-19 18:28:32,646 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
```

janiversen, 2023-07-19:
So it seems you are not "just" running server_async and client_calls as the issue describes.

Closing this as it does not seem to be a pymodbus problem.

closed 2023-07-19 by janiversen

DivineMK, 2023-07-20:
i tried cloning again but still not working. I wonder if using two different python affects which packages is installed though as that might be my issue.

janiversen, 2023-07-20:
If you cloned, you need to install the virtual environment (see README for how to install a development environment). especially if you have 2 different python versions. Mixing site_packages between version is a very bad idea, that are likely to produce some strange errors.

I have 4 different pythons installed (to do different testing), and have no problems (the output I posted is from python 3.10).

DivineMK, 2023-07-20:
```
Traceback (most recent call last):
  File "/home/mk/.local/bin/pre-commit", line 5, in <module>
    from pre_commit.main import main
  File "/home/mk/.local/lib/python3.8/site-packages/pre_commit/main.py", line 12, in <module>
    from pre_commit.commands.autoupdate import autoupdate
  File "/home/mk/.local/lib/python3.8/site-packages/pre_commit/commands/autoupdate.py", line 19, in <module>
    from pre_commit.store import Store
  File "/home/mk/.local/lib/python3.8/site-packages/pre_commit/store.py", line 6, in <module>
    import sqlite3
  File "/usr/local/lib/python3.8/sqlite3/__init__.py", line 23, in <module>
    from sqlite3.dbapi2 import *
  File "/usr/local/lib/python3.8/sqlite3/dbapi2.py", line 27, in <module>
    from _sqlite3 import *
ModuleNotFoundError: No module named '_sqlite3'
```
I got this error when running pre-commit --install. Does that mean sqlite3 is missing?

janiversen, 2023-07-20:
It means you did not follow the readme.

janiversen, 2023-07-20:
Apart from that, if you have problems with "pre-commit" ask in that project, we are the pymodbus project.

janiversen, 2023-07-20:
what do you think ? Follow the instructions in the README, that help.

DivineMK, 2023-07-20:
Oh it still works with that error. Thank you very much.