pymodbus-dev/pymodbus #1683: Connection issue with async client

alexrudd2, 2023-07-20:
### Versions

- Python: `3.11`
- OS: MacOS
- Pymodbus: `3.4.0`
- Modbus Hardware: Watlow RUI gateway
- async client

### Description

`connect()` now fails on `3.4.0` with code that worked on previous versions.

### Code and Logs

```python
# pymodbus 3.3.2
2023-07-20 16:00:47,365 DEBUG logging:102 Connecting to 192.168.10.242:502.
2023-07-20 16:00:47,365 DEBUG logging:102 Connecting comm
2023-07-20 16:00:47,367 DEBUG logging:102 Connected to comm
2023-07-20 16:00:47,367 DEBUG logging:102 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x1 0x68 0x0 0x2
2023-07-20 16:00:47,367 DEBUG logging:102 Adding transaction 1
2023-07-20 16:00:47,467 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x0 0x3 0x4 0x41 0xb9
# everything normal, further logs omitted

# pymodbus 3.4.0
2023-07-20 16:04:13,157 DEBUG logging:102 Connecting to 192.168.10.242:502.
2023-07-20 16:04:13,157 DEBUG logging:102 Connecting comm
# application times out and kills connection
```

I will attempt to bisect.

alexrudd2, 2023-07-20:
```
04117dfb17e9f5b28e8156d80b04fba658012c1d is the first bad commit
commit 04117dfb17e9f5b28e8156d80b04fba658012c1d
Author: jan iversen <jancasacondor@gmail.com>
Date:   Wed Jul 5 20:43:04 2023 +0200

    Remove self.params from async client (#1640)
```

Will attempt to bisect further on that branch.  My guess is some default parameters are not being passed correctly.  My client code uses very few explicit args:
`self.client = AsyncModbusTcpClient(address, timeout=timeout)`

alexrudd2, 2023-07-20:
I suspect the change is because `host` is now required to be a keyword argument.  Previously it worked as a positional argument.

```
    class _params:
---         host: str = None

       ModbusClientMixin.__init__(self)
+++        setup_params = CommParams(
+++            host=kwargs.get("host", None),
```

janiversen, 2023-07-21:
hmmm the defaults should be defined in the __init__ method, I just changed away from the self.params, to the new self.comm_params. The aim is to remove self.params completely.

janiversen, 2023-07-21:
I will try to make a small test case for this, I looked at the commit you pointed out and could not see anything wrong (but there still might be something).

janiversen, 2023-07-21:
hmmm it works for me (after I remembered to change the server to use port 502):
```
./mytest.py
get client
connect to server
2023-07-21 08:50:25,006 DEBUG logging:102 Connecting to 127.0.0.1:502.
2023-07-21 08:50:25,006 DEBUG logging:102 Connecting comm
2023-07-21 08:50:25,007 DEBUG logging:102 Connected to comm
2023-07-21 08:50:25,007 DEBUG logging:102 callback_connected called
get and verify data
2023-07-21 08:50:25,007 DEBUG logging:102 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x1 0x0 0x1
2023-07-21 08:50:25,008 DEBUG logging:102 Adding transaction 1
2023-07-21 08:50:25,008 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x1 addr=None
2023-07-21 08:50:25,008 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x1
2023-07-21 08:50:25,008 DEBUG logging:102 Factory Response[ReadCoilsResponse': 1]
2023-07-21 08:50:25,008 DEBUG logging:102 Getting transaction 1
close connection
```
However the problem might be source_address, so let me correct that.

PR #1690 (2023-07-21): Change default source_address to (0.0.0.0, 502)

fixes #1683
closed 2023-07-21 by janiversen