pymodbus-dev/pymodbus #1695: RTU regression in 3.3.0 for custom message class

wlcrs, 2023-07-24:
### Versions

- Python: 3.11.4
- OS: Linux
- Pymodbus: Works fine up until 3.2.2. Regression started in 3.3.0
- Modbus Hardware (if used):

### Pymodbus Specific

- Client: rtu async

### Description

I have multiple people reporting a failure to communicate over Modbus RTU using a custom message class which has `_rtu_byte_count_pos` set. When they downgrade to pyModbus 3.2.2 it starts working again. The same code also works fine for connections over Modbus TCP.

cfr. https://github.com/wlcrs/huawei_solar/issues/426 and https://github.com/wlcrs/huawei_solar/issues/422#issuecomment-1646764646

The logs show that the data arrives in two parts, and that the first part gets discarded because the frame is invalid. However, the data should be kept until the frame is complete.

### Code and Logs

## Code

```python

class AsyncHuaweiSolarModbusSerialClient(AsyncModbusSerialClient):
    """Custom SerialClient with support for custom Huawei modbus messages"""

    def __init__(self, port, baudrate, timeout: int, **serial_kwargs):
        super().__init__(port, **serial_kwargs, baudrate=baudrate, reconnect_delay=RECONNECT_DELAY, timeout=timeout)
        super().register(PrivateHuaweiModbusResponse)

...

class PrivateHuaweiModbusResponse(ModbusResponse):
    """Response with the private Huawei Solar function code"""

    function_code = 0x41
    _rtu_byte_count_pos = 3

    def __init__(self, **kwargs):
        ModbusResponse.__init__(self, **kwargs)

        self.sub_command = None
        self.content = b""

    def decode(self, data):
        self.sub_command = int(data[0])
        self.content = data[1:]

    def __str__(self):
        return f"{self.__class__.__name__}({self.sub_command})"

...

class StartUploadModbusRequest(ModbusRequest):
    """
    Modbus file upload request
    """

    function_code = 0x41
    sub_function_code = 0x05

    def __init__(self, file_type, customized_data: t.Optional[bytes] = None, **kwargs):
        ModbusRequest.__init__(self, **kwargs)
        self.file_type = file_type

        if customized_data is None:
            self.customised_data = b""
        else:
            self.customised_data = customized_data

    def encode(self):
        data_length = 1 + len(self.customised_data)
        return struct.pack(">BBB", self.sub_function_code, data_length, self.file_type) + self.customised_data

    def decode(self, data):
        sub_function_code, data_length, self.file_type = struct.unpack(">BBB", data)
        self.customised_data = data[3:]

        assert sub_function_code == self.sub_function_code
        assert len(self.customised_data) == data_length - 1

```

## Logs
```
2023-07-23 10:59:47.460 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x45 from slave 1
2023-07-23 10:59:47.460 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x41 0x5 0x1 0x45 0xdd 0x9e
2023-07-23 10:59:47.460 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] Frame check failed, ignoring!!
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x01A\x05\nE\x00\x00\x07N\xf6'] not ready
2023-07-23 10:59:47.550 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-23 10:59:47.550 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-23 10:59:47.551 DEBUG (MainThread) [pymodbus.logging] Frame check failed, ignoring!!
2023-07-23 10:59:47.551 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 0x0 0x7 0x4e 0xf6 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-23 10:59:57.462 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (TimeoutError)
```

janiversen, 2023-07-24:
Please add the same log, but for 3.2.2 where it works.

The reason is the code around "Frame check failed, ignoring!!" have not been changed so in 3.2.2 it would also reset the frame.

I have no way of easily simulating your code.

janiversen, 2023-07-24:
another thought, if it comes a 2 packets on a serial line with a pause in between, it is to be interpreted as 2 frames, that is the definition of rtu framer. That could be the whole problem, especially if someone uses the "strict" parameter.

wlcrs, 2023-07-24:
Hi Jan,

The user was using 3.3.2 to begin with. I initially asked the user to upgrade to pymodbus 3.4.0 as I indeed saw some RTU-specific fixes in the changelog. This did not fix it for him. Downgrading to pymodbus 3.2.2 did.

I'll ask for some logs using 3.2.2 from @julianbl.

wlcrs, 2023-07-24:
The user provided the following logs showing two messages, one short followed by one very long message:

```
2023-07-24 12:28:35.339 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2260273205_configuration_update_coordinator data in 0.889 seconds (success: True)
2023-07-24 12:28:35.399 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x45 from slave 1
2023-07-24 12:28:35.400 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x41 0x5 0x1 0x45 0xdd 0x9e
2023-07-24 12:28:35.402 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-24 12:28:35.536 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x41 0x5 0xa
2023-07-24 12:28:35.536 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x01A\x05\n'] not ready
2023-07-24 12:28:35.552 DEBUG (MainThread) [pymodbus.logging] recv: 0x45 0x0 0x0 0x7 0x4e 0xf6 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-24 12:28:35.552 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6 0xff 0xff 0xff 0xff
2023-07-24 12:28:35.552 DEBUG (MainThread) [pymodbus.logging] Factory Response[PrivateHuaweiModbusResponse': 65]
2023-07-24 12:28:35.553 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-07-24 12:28:35.553 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-07-24 12:28:35.553 DEBUG (MainThread) [pymodbus.logging] Frame - [b'E\x00\x00\x07N\xf6\xff\xff\xff\xff\xd8>'] not ready
2023-07-24 12:28:35.553 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x41 0x6 0x3 0x45 0x0 0x0 0x9d 0x40
2023-07-24 12:28:35.553 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-24 12:28:35.586 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x41 0x6 0xf9 0x45 0x0
2023-07-24 12:28:35.587 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x01A\x06\xf9E\x00'] not ready
2023-07-24 12:28:35.603 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x56 0x31 0x30 0x33 0x8 0x7e 0x7 0x46 0x0 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.603 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00V103\x08~\x07F\x00\x00\x00\x00\x00'] not ready
2023-07-24 12:28:35.620 DEBUG (MainThread) [pymodbus.logging] recv: 0x10 0x0 0x4 0x0 0x1 0x2 0x2 0xff 0xff 0x4e 0x53 0x32 0x32 0x38
2023-07-24 12:28:35.620 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x10\x00\x04\x00\x01\x02\x02\xff\xffNS228'] not ready
2023-07-24 12:28:35.632 DEBUG (MainThread) [pymodbus.logging] recv: 0x39 0x31 0x32 0x33 0x36 0x38 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.632 DEBUG (MainThread) [pymodbus.logging] Frame - [b'9123681\x00\x00\x00\x00\x00\x00\x00'] not ready
2023-07-24 12:28:35.648 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x32 0x43 0x31 0x30 0x53 0x50
2023-07-24 12:28:35.648 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00V100R002C10SP'] not ready
2023-07-24 12:28:35.664 DEBUG (MainThread) [pymodbus.logging] recv: 0x43 0x30 0x31 0x39 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.664 DEBUG (MainThread) [pymodbus.logging] Frame - [b'C019\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'] not ready
2023-07-24 12:28:35.680 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.680 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'] not ready
2023-07-24 12:28:35.696 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x55 0x4e 0x32 0x30
2023-07-24 12:28:35.696 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00\x00\x00\x00\x00\x00\x00\x00\x00SUN20'] not ready
2023-07-24 12:28:35.712 DEBUG (MainThread) [pymodbus.logging] recv: 0x30 0x30 0x2d 0x34 0x35 0x30 0x57 0x2d 0x50 0x32 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.712 DEBUG (MainThread) [pymodbus.logging] Frame - [b'00-450W-P2\x00\x00\x00\x00'] not ready
2023-07-24 12:28:35.728 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x8 0x0
2023-07-24 12:28:35.728 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00'] not ready
2023-07-24 12:28:35.744 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x1 0xc2 0x0 0x1 0x0 0x5 0x0 0x1 0x2 0x2 0xff 0xff 0x4e
2023-07-24 12:28:35.744 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x01\x01\xc2\x00\x01\x00\x05\x00\x01\x02\x02\xff\xffN'] not ready
2023-07-24 12:28:35.760 DEBUG (MainThread) [pymodbus.logging] recv: 0x53 0x32 0x32 0x38 0x39 0x31 0x32 0x33 0x37 0x30 0x35 0x0 0x0 0x0
2023-07-24 12:28:35.760 DEBUG (MainThread) [pymodbus.logging] Frame - [b'S2289123705\x00\x00\x00'] not ready
2023-07-24 12:28:35.776 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x32
2023-07-24 12:28:35.776 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00\x00\x00\x00\x00V100R002'] not ready
2023-07-24 12:28:35.792 DEBUG (MainThread) [pymodbus.logging] recv: 0x43 0x31 0x30 0x53 0x50 0x43 0x30 0x31 0x39 0x0 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.792 DEBUG (MainThread) [pymodbus.logging] Frame - [b'C10SPC019\x00\x00\x00\x00\x00'] not ready
2023-07-24 12:28:35.808 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-07-24 12:28:35.808 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'] not ready
[... cut ...] x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x34 0x35 0x30 0x57 0x2d 0x50 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x1 0x1 0xc2 0x0 0x1
2023-07-24 12:28:35.873 DEBUG (MainThread) [pymodbus.logging] Factory Response[PrivateHuaweiModbusResponse': 65]
2023-07-24 12:28:35.873 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-07-24 12:28:35.873 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
```

janiversen, 2023-07-26:
This is working on dev:
```
2023-07-26 20:20:58,309 DEBUG logging:102 Connecting to /dev/ttys006.
2023-07-26 20:20:58,309 DEBUG logging:102 Connecting comm
2023-07-26 20:20:58,310 DEBUG logging:102 Connected to comm
2023-07-26 20:20:58,310 DEBUG logging:102 send: 0x1 0x1 0x0 0x1 0x0 0x50 0x6d 0xf6
2023-07-26 20:20:58,310 DEBUG logging:102 Adding transaction 1
2023-07-26 20:20:58,311 DEBUG logging:102 recv: 0x1 0x1 0xa 0xff 0xff
2023-07-26 20:20:58,311 DEBUG logging:102 Processing: 0x1 0x1 0xa 0xff 0xff
2023-07-26 20:20:58,311 DEBUG logging:102 Frame - not ready
2023-07-26 20:20:58,311 DEBUG logging:102 recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xec 0xf2
2023-07-26 20:20:58,312 DEBUG logging:102 Processing: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xec 0xf2
2023-07-26 20:20:58,312 DEBUG logging:102 Getting Frame - 0x1 0xa 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-07-26 20:20:58,312 DEBUG logging:102 Factory Response[ReadCoilsResponse': 1]
2023-07-26 20:20:58,312 DEBUG logging:102 Frame advanced, resetting header!!
2023-07-26 20:20:58,312 DEBUG logging:102 Getting transaction 1
```

I modified the async serial server to send the response frame as 2 packets, with a flush() and a short delay in between.

This will be part of 3.4.1 which are due very soon.

wlcrs, 2023-07-31:
The user reported that the issue isn't resolved in v3.4.1 . I've been tinkering with the pyModbus code to replicate the issue when I noticed the following: