pymodbus-dev/pymodbus #1697: Asynchronous RTU serial server not detecting port string

jannikgoebel, 2023-07-25:
### Versions

- Python: 3.11.4
- OS: Windows
- Pymodbus: Works with 3.3.2 Stop working with 3.4.0
- Modbus Hardware (if used): FTDI Based RS485<->USB Adapter

### Pymodbus Specific

- Server: rtu- async

### Description

With the new Version it is not possible to start a server, because the Port String is not detected and the default Value of 127.0.0.1 is used.

### Code and Log

## Code
```python
import asyncio
import logging

from pymodbus import __version__ as pymodbus_version
from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusServerContext,
    ModbusSlaveContext,
    ModbusSparseDataBlock,
)
from pymodbus.device import ModbusDeviceIdentification

import argparse
import os

from pymodbus import pymodbus_apply_logging_config
from pymodbus.transaction import ModbusRtuFramer
from pymodbus.server import StartAsyncSerialServer
pymodbus_apply_logging_config()
logging.basicConfig()
_logger = logging.getLogger(__file__)
_logger.setLevel(logging.DEBUG)

async def run_async_server():
    """Run server."""
    datablock = ModbusSequentialDataBlock.create()
    server = await StartAsyncSerialServer(
        context=ModbusServerContext(
            slaves={
                0x01: ModbusSlaveContext(
                    di=datablock,
                    co=datablock,
                    hr=datablock,
                    ir=datablock,
                ),
                0x02: ModbusSlaveContext(
                    di=datablock,
                    co=datablock,
                    hr=datablock,
                    ir=datablock,
                ),
            },
            single=False
        ),
                # Data storage
        identity=ModbusDeviceIdentification(
            info_name={
                "VendorName": "Pymodbus",
                "ProductCode": "PM",
                "VendorUrl": "https://github.com/pymodbus-dev/pymodbus/",
                "ProductName": "Pymodbus Server",
                "ModelName": "Pymodbus Server",
                "MajorMinorRevision": pymodbus_version,
            }
        ),  # server identify
        port="COM11",  # serial port
        framer=ModbusRtuFramer,  # The framer strategy to use
        baudrate=19200,  # The baud rate to use for the serial device
    )    
    return server

async def async_helper():
    _logger.info("Starting...")
    await run_async_server()

if __name__ == "__main__":
    asyncio.run(async_helper(), debug=True)  # pragma: no cover
```

## Log
```txt
INFO:C:\Users\jannik.goebel\Documents\Projects\python\modbusServer\example.py:Starting...
2023-07-25 10:12:35,621 INFO  logging:96 Server(Serial) listening.
INFO:pymodbus.logging:Server(Serial) listening.
2023-07-25 10:12:35,622 DEBUG logging:102 Awaiting connections server_listener
DEBUG:pymodbus.logging:Awaiting connections server_listener
2023-07-25 10:12:35,622 WARNING logging:108 Failed to start server could not open port '127.0.0.1': FileNotFoundError(2, 'Das System kann die angegebene Datei nicht finden.', None, 2)
WARNING:pymodbus.logging:Failed to start server could not open port '127.0.0.1': FileNotFoundError(2, 'Das System kann die angegebene Datei nicht finden.', None, 2)
2023-07-25 10:12:35,622 ERROR logging:114 Server unexpected exception 'NoneType' object has no attribute 'serve_forever'
ERROR:pymodbus.logging:Server unexpected exception 'NoneType' object has no attribute 'serve_forever'
```

janiversen, 2023-07-25:
Hmmm I am pretty sure we check that, but let me investigate.

Apart from that your port string seems wrong, you need to point at the device driver.

janiversen, 2023-07-25:
Please test with current dev and verify it works. You might be able to just apply the bug fix, but there are other changes and I have not controlled which depend on what.

Please let us know if everything works, we are planning to release 3.4.1 within a week, and this patch with be part if confirmed.

jannikgoebel, 2023-07-26:
I have tried both variants. Unfortunately, both the dev branch and a manual bug fix where I only added the changes to transport.py is not runnable. 

### Dev Branch
```log
(env) C:\Users\jannik.goebel\Documents\Projects\python\modbusServer>python example.py
INFO:C:\Users\jannik.goebel\Documents\Projects\python\modbusServer\example.py:Starting...
2023-07-26 09:31:39,151 INFO  logging:96 Server(Serial) listening.
INFO:pymodbus.logging:Server(Serial) listening.
2023-07-26 09:31:39,151 DEBUG logging:102 Awaiting connections server_listener
DEBUG:pymodbus.logging:Awaiting connections server_listener
2023-07-26 09:31:39,163 ERROR logging:114 Server unexpected exception 'SerialTransport' object has no attribute 'serve_forever'
ERROR:pymodbus.logging:Server unexpected exception 'SerialTransport' object has no attribute 'serve_forever'
2023-07-26 09:31:39,168 DEBUG logging:102 Connected to server
DEBUG:pymodbus.logging:Connected to server
ERROR:asyncio:Exception in callback SerialTransport._ensure_reader()
handle: <Handle SerialTransport._ensure_reader() created at C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\transport\transport_serial.py:31>
source_traceback: Object created at (most recent call last):
  File "C:\Users\jannik.goebel\Documents\Projects\python\modbusServer\example.py", line 65, in async_helper
    await run_async_server()
  File "C:\Users\jannik.goebel\Documents\Projects\python\modbusServer\example.py", line 27, in run_async_server
    server = await StartAsyncSerialServer(
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\server\async_io.py", line 778, in StartAsyncSerialServer
    await _serverList.run(server, custom_functions)
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\server\async_io.py", line 651, in run
    await server.serve_forever()
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\server\async_io.py", line 617, in serve_forever
    await self.transport_listen()
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\transport\transport.py", line 249, in transport_listen
    self.transport = await self.call_create()
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\transport\transport_serial.py", line 229, in create_serial_connection
    transport = SerialTransport(loop, protocol, *args, **kwargs)
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\transport\transport_serial.py", line 31, in __init__
    loop.call_soon(self._ensure_reader)
Traceback (most recent call last):
  File "C:\Users\jannik.goebel\scoop\apps\python\current\Lib\asyncio\events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\jannik.goebel\Documents\Projects\python\pymodbus-dev\pymodbus\pymodbus\transport\transport_serial.py", line 160, in _ensure_reader
    self._has_reader = self._protocol.loop.call_later(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'call_later'
```

### Bugfix in Transport.py

```log
(bugfix) C:\Users\jannik.goebel\Documents\Projects\python\modbusServer>python example.py
INFO:C:\Users\jannik.goebel\Documents\Projects\python\modbusServer\example.py:Starting...
2023-07-26 10:11:35,551 INFO  logging:96 Server(Serial) listening.
INFO:pymodbus.logging:Server(Serial) listening.
2023-07-26 10:11:35,551 DEBUG logging:102 Awaiting connections server_listener
DEBUG:pymodbus.logging:Awaiting connections server_listener
2023-07-26 10:11:35,557 ERROR logging:114 Server unexpected exception 'SerialTransport' object has no attribute 'serve_forever'
ERROR:pymodbus.logging:Server unexpected exception 'SerialTransport' object has no attribute 'serve_forever'
2023-07-26 10:11:35,558 DEBUG logging:102 Connected to server
DEBUG:pymodbus.logging:Connected to server
```

### Working Version 3.3.2

```log
(v332) C:\Users\jannik.goebel\Documents\Projects\python\modbusServer>python example.py
INFO:C:\Users\jannik.goebel\Documents\Projects\python\modbusServer\example.py:Starting...
2023-07-26 10:12:49,887 INFO  logging:96 Server(Serial) listening.
INFO:pymodbus.logging:Server(Serial) listening.
2023-07-26 10:12:49,890 DEBUG logging:102 Serial connection opened on port: COM50
DEBUG:pymodbus.logging:Serial connection opened on port: COM50
2023-07-26 10:12:49,891 DEBUG logging:102 Serial connection established
DEBUG:pymodbus.logging:Serial connection established
```

janiversen, 2023-07-26:
thanks, this is another bug.

janiversen, 2023-07-26:
Dev branch work now in my local setup, please give it a try.

jannikgoebel, 2023-07-27:
The new Dev Version is working on my Windows now, too.
Thanks for the quick Fix.

PR #1699 (2023-07-25): Use source_address in serial.

fixes #1697
closed 2023-07-27 by jannikgoebel