pymodbus-dev/pymodbus #1709: `allow_reuse_address` missing argument on TCP server

TheUbuntuGuy, 2023-07-27:
### Versions

- Python: 3.9
- OS: Debian 11 Bullseye
- Pymodbus: 3.4.1

### Pymodbus Specific

- Server: tcp
- Client: tcp

### Description

With the release of v3.4.0, the `allow_reuse_address` argument to `StartTcpServer()` no longer exists.

This was removed from UDP in https://github.com/pymodbus-dev/pymodbus/pull/1246, however this broke TCP as well.

### Code and Logs

```python
# used to work
from multiprocessing import Process
from pymodbus.server import StartTcpServer

    kwargs = {
        "context": context,
        "address": ("0.0.0.0", port),
        "allow_reuse_address": True
    }

    process = Process(target=StartTcpServer, kwargs=kwargs, daemon=True)
```

Now yields:
```
File "/usr/local/lib/python3.9/dist-packages/pymodbus/server/async_io.py", line 788, in StartTcpServer
    return asyncio.run(StartAsyncTcpServer(**kwargs))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.9/dist-packages/pymodbus/server/async_io.py", line 692, in StartAsyncTcpServer
    server = ModbusTcpServer(
TypeError: __init__() got an unexpected keyword argument 'allow_reuse_address'
```

janiversen, 2023-07-27:
Please read API_CHANGES.rst, this is a documented change and not a bug.

closed 2023-07-27 by janiversen

TheUbuntuGuy, 2023-07-27:
Sorry, I was confused as you wrote in https://github.com/pymodbus-dev/pymodbus/pull/1246#pullrequestreview-1241997629:
> Removing reuse_addr for UDP is a good thing, I thought I had done that already. **BUT you cannot remove reuse_addr from TCP, this is highly needed.**

which is why I assumed it was to be kept in that case. Is there simply going to be no way to have this option in the future? This affects servers that restart and need port rebinding.

janiversen, 2023-07-27:
Pull requests are welcome.

janiversen, 2023-07-28:
Do you have an actual problem? If so please show the debug log, the library works different now as it did 7month ago, so the option is no longer needed in our opinion.

TheUbuntuGuy, 2023-07-28:
I took some time to read through the code and I now see how this option is set by default in the TCP transport. I was recalling how things worked a few versions ago and I agree that the option is no longer needed. Thanks