pymodbus-dev/pymodbus #1711: pymodbus.server does not start in 3.4.1

chkr-private, 2023-07-30:
### Versions

- Python: 3.11.4
- OS: Fedora 37
- Pymodbus: 3.4.1
- Modbus Hardware (if used): None

### Pymodbus Specific

- Server: tcp

### Description

- "pymodbus.server run" does not start

Error message:
```pymodbus.server run
Traceback (most recent call last):

  File "/usr/bin/pymodbus.server", line 8, in <module>
    sys.exit(app())
             ^^^^^

  File "/usr/lib/python3.11/site-packages/pymodbus/repl/server/main.py", line 190, in run
    app = ReactiveServer.factory(
          ^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3.11/site-packages/pymodbus/server/reactive/main.py", line 505, in factory
    server = server(
             ^^^^^^^

TypeError: ModbusTcpServer.__init__() got an unexpected keyword argument 'loop'
```

### Code and Logs
As a workaround, the following patch for class ModbusTcpServer helps but I'm unsure whether the "loop" parameter is actually needed or not.

```
--- server/async_io.py.old	2023-07-30 11:11:19.538544170 +0200
+++ server/async_io.py	2023-07-30 11:11:32.463766887 +0200
@@ -269,6 +269,7 @@
         broadcast_enable=False,
         response_manipulator=None,
         request_tracer=None,
+        **kwargs,
     ):
         """Initialize the socket server.
 ```

The problem looks similar to #1662, but this time with the REPL stand-alone tool pymodbus.server.

janiversen, 2023-07-30:
It works as a workaround, but clearly the loop is not used as parameter and adding warns just hides the problem.

The caller should not add loop, that is the correct solution.

Feel free to submit a PR and I will review it.

chkr-private, 2023-07-30:
> It works as a workaround, but clearly the loop is not used as parameter and adding warns just hides the problem.

Unfortunately, there were additional superfluous parameters.

> Feel free to submit a PR and I will review it.

Please see #1714.

janiversen, 2023-07-31:
You are right, the main problem is that there are no tests trying to start the REPL server so problems are not detected.

janiversen, 2023-08-13:
This works now on dev.

closed 2023-08-13 by janiversen

chkr-private, 2023-08-13:
> This works now on dev.

I tested dev at commit 08529612b11966e4d16d085b7550d74737472b67 and the problem is still reproducible. It looks like that #1714 is not merged yet.