pymodbus-dev/pymodbus #1712: pymodbus.server does not work with serial port

chkr-private, 2023-07-30:
### Versions

- Python: 3.11.4
- OS: Fedora 37
- Pymodbus: 3.4.1
- Modbus Hardware (if used): none (test was done with a pair of pseudo-terminals)

### Pymodbus Specific

- Server: rtu/serial

### Description

- it is not possible to specify the serial device for "pymodbus.server run -s serial"
- `pymodbus.server run -s serial -p ./device/p1` 
```
pymodbus.server run -s serial -p ./device/p1
Usage: pymodbus.server run [OPTIONS]
Try 'pymodbus.server run --help' for help.

Error: Invalid value for '--modbus-port' / '-p': './device/p1' is not a valid integer.
```

- the problem seems to be, that the parameter "modbus_port" is used for both, the TCP port as well as for the serial device and defined to require an "int" in typer
- the following patch work for both cases (TCP port number as well as serial device)
```
--- repl/server/main.py.old	2023-07-30 11:29:22.987247743 +0200
+++ repl/server/main.py	2023-07-30 11:29:31.883401640 +0200
@@ -137,7 +137,7 @@
         autocompletion=framers,
         help="Modbus framer to use",
     ),
-    modbus_port: int = typer.Option(5020, "--modbus-port", "-p", help="Modbus port"),
+    modbus_port: str = typer.Option(5020, "--modbus-port", "-p", help="Modbus port"),
     modbus_slave_id: List[int] = typer.Option(
         [1], "--slave-id", "-u", help="Supported Modbus slave id's"
     ),
```
- if that that's the correct way of fixing it, please let me know and I'll create a pull request

janiversen, 2023-07-30:
Sounds correct, please. submit a PR and let CI give a verdict.

alexrudd2, 2023-07-31:
Closed by https://github.com/pymodbus-dev/pymodbus/pull/1713.

I'm not sure why the "Fixes XYZ" are not automatically linking the issues.

closed 2023-07-31 by alexrudd2