pymodbus-dev/pymodbus #1722: BUG: handle_local_echo not working

Iktek, 2023-08-07:
### Discussed in https://github.com/pymodbus-dev/pymodbus/discussions/1721

<div type='discussions-op-text'>

<sup>Originally posted by **Iktek** August  7, 2023</sup>
Dear List, 
On current dev version there seems to be a bug with "handle_local_echo".
The attribute is pushed to the base-clients "comm_params" variable in base.py.

Transaction.py nevertheless tries to find it in the base of the client class:

hasattr(self.client, "handle_local_echo")

This seems wrong.
When I change transaction.py like:

```
--- transaction.py.old  2023-08-07 09:50:09.485911612 +0200
+++ transaction.py      2023-08-07 09:42:42.276994370 +0200
@@ -311,8 +311,8 @@ class ModbusTransactionManager:
                 )
                 self.client.state = ModbusTransactionState.WAITING_FOR_REPLY
             if (
-                hasattr(self.client, "handle_local_echo")
-                and self.client.handle_local_echo is True
+                hasattr(self.client.comm_params, "handle_local_echo")
+                and self.client.comm_params.handle_local_echo is True
             ):
                 if self._recv(size, full) != packet:
                     return b"", "Wrong local echo"

```

The functionality seems to work again.

</div>

janiversen, 2023-08-07:
Looks like a bug, please submit a pull request

janiversen, 2023-08-07:
2 comments
- for async clients local_echo is handled in transport
- comm_params is a dataclass so the parameter is always present
- the code cannot work it misses an "or" to combine the 2 conditions

Iktek, 2023-08-07:
Hi JanIversen, 
just to clarify your points:
- transport class does not have this bug it seems
- so you mean that the "hasattr" is not needed here.. 
- which 2 conditions? I think they are "and" ed together as before or do you mean to leave the previous check in for compatibility reasons?
Regards Pascal

janiversen, 2023-08-07:
* no it does not
* yes
* I see your -- and ++ that makes the difference, if you are sure the other part is not used (of which I am not so sure).

janiversen, 2023-08-13:
Still looking for a Pull request.

janiversen, 2023-08-13:
This is working on dev (you are comparing old code).

closed 2023-08-13 by janiversen