pymodbus-dev/pymodbus #1724: Simple server / client example that is actually able to be followed

MathijsNL, 2023-08-07:
### Description

As in the title. It is really hard to run a pymodbus server and just print some data to get started. I have no clue if my PLC connects, because the server examples by default spam the console so fast. This is happening with every server example.

I think it would be good if a server + client example could be added where the client just sends a few bits every x seconds and the server receives and prints them. This way people can actually connect some other hardware and connect to the server example and see things happening.

```
DEBUG:__main__:Callback from validate with address 1, count 1, data True
DEBUG:__main__:Callback from setValues with address 1, value [True]
DEBUG:__main__:Callback from getValues with address 1, count 1, data [True]
DEBUG:__main__:Callback from validate with address 1, count 1, data True
DEBUG:__main__:Callback from setValues with address 1, value [True]
DEBUG:__main__:Callback from getValues with address 1, count 1, data [True]
DEBUG:__main__:Callback from validate with address 1, count 1, data True
DEBUG:__main__:Callback from setValues with address 1, value [True]
DEBUG:__main__:Callback from getValues with address 1, count 1, data [True]
DEBUG:__main__:Callback from validate with address 1, count 1, data True
DEBUG:__main__:Callback from setValues with address 1, value [False]
DEBUG:__main__:Callback from getValues with address 1, count 1, data [False]
DEBUG:__main__:Callback from validate with address 1, count 1, data True
DEBUG:__main__:Callback from setValues with address 1, value [False]
DEBUG:__main__:Callback from getValues with address 1, count 1, data [False]
DEBUG:__main__:Callback from validate with address 1, count 1, data True
DEBUG:__main__:Callback from setValues with address 1, value [False]
DEBUG:__main__:Callback from getValues with address 1, count 1, data [False]
```

janiversen, 2023-08-07:
We have a lot of examples that show client/server without flooding the log. For example use the simple_client together with a server, that produces around 30 log lines.

Having said that pull requests are more than welcome, if you want to make better examples.

Btw. Seems you have enabled debug on the server side otherwise you would see a lot less log lines.

closed 2023-08-07 by janiversen

MathijsNL, 2023-08-07:
Thanks for your response.

I think my point is, if a bare minimum example to connect a client to a server is hundreds of lines of code it is going to be really hard for people to get going with modbus.

In the meantime I found an example that is exactly what I was looking for.

[Example from APMonitor](https://apmonitor.com/dde/index.php/Main/ModbusTransfer)

I think it is not appropriate if I make a PR with the code from APMonitor, I will send him a message so he can decide on what to do but I think it would be nice if his example would be in the contrib examples.

janiversen, 2023-08-07:
As you please....our examples try to show how to use especially the client in a simple way (see simple_client) but we show the different communication types and different modbus frames as simple options. A client is as simple as 3 api calls, and a server can be done in 5 calls.

The example you refer to, claims to be async ("run_async_server()") but the function is not even declared async, that is the problems we see. What would you do with that example if you wanted serial communication, just asking.

Having said all that we are happy that you have found the help you need, good luck in making your application.