pymodbus-dev/pymodbus #1735: data is not correct

zouzhe1, 2023-08-14:
### Versions

- Python:3.11.4
- OS:win10
- Pymodbus:   laster
- Modbus Hardware (if used):      serial 485    and    virtual com

### Pymodbus Specific

- Client: rtu - sync

### Description

i want send 【01 10 60 FE 00 02 04 00 00 00 01 14 A5】,but pymodbus send 【01 10 60 FE 00 04 08 00 02 04 00 00 00 01 00 BE D8 】

### Code and Logs

```python
# code and logs here.

from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.client import ModbusSerialClient
from pymodbus import pymodbus_apply_logging_config
pymodbus_apply_logging_config()
client = ModbusSerialClient(
    method="rtu",
    port="COM2",
    stopbits=1,
    bytesize=8,
    parity='N',
    baudrate=115200
)
client.connect()

# 创建BinaryPayloadBuilder对象
builder = BinaryPayloadBuilder(wordorder=Endian.Big, byteorder=Endian.Big)

data32 = 0x00000001

builder.add_16bit_uint(0x0002)
builder.add_8bit_uint(0x04)
builder.add_32bit_uint(data32)

# 获取寄存器数据
registers = builder.to_registers()

# 写入数据到寄存器
client.write_registers(0x60FE, registers, slave=1)

# 关闭连接
client.close()

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config()
```
2023-08-14 09:55:31,924 DEBUG logging:102 [2, 1024, 0, 256]
2023-08-14 09:55:31,924 DEBUG logging:102 Current transaction state - IDLE
2023-08-14 09:55:31,925 DEBUG logging:102 Running transaction 1
2023-08-14 09:55:31,925 DEBUG logging:102 SEND: 0x1 0x10 0x60 0xfe 0x0 0x4 0x8 0x0 0x2 0x4 0x0 0x0 0x0 0x1 0x0 0xbe 0xd8
2023-08-14 09:55:31,925 DEBUG logging:102 New Transaction state "SENDING"
2023-08-14 09:55:31,926 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"

janiversen, 2023-08-14:
Registers in modbus are 16bit, but you want to send 7bytes that will not work.

You also did not specify count= in the write_registers call.

janiversen, 2023-08-14:
It sends correctly what you have defined, but I think your payload builder call do not correspond to what you want to send, please see the modbus specification. We provide a link to the official documentation in our README.rst

janiversen, 2023-08-14:
Closing as it is not a pymodbus bug.

closed 2023-08-14 by janiversen