pymodbus-dev/pymodbus #1740: Pymodbus does not connect

mukesh-rudrasec, 2023-08-19:
### Versions

- Python: 3.11.4
- OS: Windows and Linux (Mac not tested)
- Pymodbus: 3.4.x
- Modbus Hardware (if used): Simulated using [DTM](https://www.trianglemicroworks.com/products/testing-and-configuration-tools/dtm-pages/overview)

### Pymodbus Specific

- Client: tcp - sync

### Description

We were trying to connect to a simulated Modbus device and the connection was throwing an exception. The connection works perfectly when using version 3.3.2, but doesn't work when using version 3.4.x

### Code and Logs

```
Traceback (most recent call last):
  File "/home/mukesh/Downloads/script/script.py", line 65, in <module>
    main()
  File "/home/mukesh/Downloads/script/script.py", line 57, in main
    response = scan_device(client)
               ^^^^^^^^^^^^^^^^^^^
  File "/home/mukesh/Downloads/script/script.py", line 21, in scan_device
    response = client.read_device_information(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mukesh/.local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 502, in read_device_information
    return self.execute(
           ^^^^^^^^^^^^^
  File "/home/mukesh/.local/lib/python3.11/site-packages/pymodbus/client/base.py", line 175, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.123:502)]
```

mukesh-rudrasec, 2023-08-19:
cc @KartikSoneji

janiversen, 2023-08-19:
Did you call connect(), without seeing your code we have no way of helping you.

Please make an example as short as possible that shows your problem, that enables us to verify your problem.

Furthermore please activate the pymodbus debug logging, to see what is actually happening.

Between v3.3 and v3.4 are api changes as documented in API_CHANGES which might cause changes in your code.

Apart from that the exception is strange to me, and not what pymodbus normally throws.

KartikSoneji, 2023-08-19:
Hi @janiversen , thanks for the quick reply

Here's a minimum reproducible example:
```python3
import pymodbus
from pymodbus.client import ModbusTcpClient as ModbusClient
pymodbus.pymodbus_apply_logging_config()

print(f"pymodbus version: v{pymodbus.__version__}")
print()
client = ModbusClient("127.0.0.1", port="502")
bits = client.read_coils(0, slave=1).bits
print()
print(f"bits: {list(map(int, bits))}")
```

And here are the respective outputs with the same server:

## v3.3.2
```console
> py script.py

pymodbus version: v3.3.2

2023-08-19 23:55:35,990 DEBUG logging:102 Connection to Modbus server established. Socket ('127.0.0.1', 58408)
2023-08-19 23:55:35,990 DEBUG logging:102 Current transaction state - IDLE
2023-08-19 23:55:35,990 DEBUG logging:102 Running transaction 1
2023-08-19 23:55:35,990 DEBUG logging:102 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x0 0x0 0x1
2023-08-19 23:55:35,990 DEBUG logging:102 New Transaction state "SENDING"
2023-08-19 23:55:35,990 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-08-19 23:55:36,011 DEBUG logging:102 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-08-19 23:55:36,012 DEBUG logging:102 RECV: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x0
2023-08-19 23:55:36,012 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x0
2023-08-19 23:55:36,012 DEBUG logging:102 Factory Response[ReadCoilsResponse': 1]
2023-08-19 23:55:36,012 DEBUG logging:102 Adding transaction 1
2023-08-19 23:55:36,013 DEBUG logging:102 Getting transaction 1
2023-08-19 23:55:36,013 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

bits: [0, 0, 0, 0, 0, 0, 0, 0]
```

## v3.4.1
```console
> py script.py

pymodbus version: v3.4.1

Traceback (most recent call last):
  File "...\script.py", line 8, in <module>
    bits = client.read_coils(0, slave=1).bits
  File "...\lib\site-packages\pymodbus\client\mixin.py", line 72, in read_coils
    return self.execute(
  File "...\lib\site-packages\pymodbus\client\base.py", line 179, in execute
    raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(127.0.0.1:502)]
```

janiversen, 2023-08-19:
You miss the .connect() call, which is why you see the connect error in the read call, which actually might work a bit differently.

the read call sometimes returns errors so the .bits is really dangerous.

Seems to me that you have not activated the pymodbus debug log (see pymodbus_apply... in the v3.4.1 log

Those are just advice if you want stable code, your code looks ok....but I suspect you miss parameters in the init call, like e.g. a longer timeout, we have a lot of parameters to allow for different devices.

If you look at our example simple_sync_client.py that might give you an idea on your problem, that one works out of the box, because we test it with every commit.

The sync client have not had any major changes between v3.3 and v3.4 but the parameter defaults have changed, so it is recommended that you read the documentation.

KartikSoneji, 2023-08-19:
> You miss the .connect() call, which is why you see the connect error in the read call, which actually might work a bit differently.

Ah I see.
Calling `.connect()` on the client object does indeed allow the script to work properly.
I was able to solve the issue, thank you.

Would you consider adding a warning message so it is easier for other people to identify the issue?
Something similar to:
https://github.com/pymodbus-dev/pymodbus/blob/cbbdf0f4b2f0102c4c1b98419cca3ac8e2980b76/pymodbus/client/base.py#L176-L185
```diff
diff --git a/pymodbus/client/base.py b/pymodbus/client/base.py
index a99d247..404516c 100644
--- a/pymodbus/client/base.py
+++ b/pymodbus/client/base.py
@@ -176,10 +176,16 @@ class ModbusBaseClient(ModbusClientMixin, ModbusProtocol):
         """
         if self.use_sync:
             if not self.connected:
-                raise ConnectionException(f"Failed to connect[{self!s}]")
+                raise ConnectionException(
+                    f"Failed to connect[{self!s}]\n"
+                    "Did you forget to call the `.connect()` method?"
+                )
             return self.transaction.execute(request)
         if not self.transport:
-            raise ConnectionException(f"Not connected[{self!s}]")
+            raise ConnectionException(
+                f"Not connected[{self!s}]\n"
+                "Did you forget to call the `.connect()` method?"
+            )
         return self.async_execute(request)
 
     # ----------------------------------------------------------------------- #
```

<br/>

> the read call sometimes returns errors so the .bits is really dangerous.

I understand, this was just a minimal example to demonstrate the error.
<br/>
> Seems to me that you have not activated the pymodbus debug log (see pymodbus_apply... in the v3.4.1 log

I ran the exact same script both times.
It seems like this error occurs before pymodbus can start logging.
<br/>

> Those are just advice if you want stable code, your code looks ok....but I suspect you miss parameters in the init call, like e.g. a longer timeout, we have a lot of parameters to allow for different devices.

Oh are there any defaults you would recommend changing?

janiversen, 2023-08-19:
v3.4.1 do not have default turned on by default, therefore you need to activate pymodbus debug.

We cannot issue warnings for people not using he library correctly. We suply examples so that people can see how to use it.

since you have solved your problem, no need to continue.

closed 2023-08-19 by janiversen

KartikSoneji, 2023-08-19:
> We cannot issue warnings for people not using he library correctly. We suply examples so that people can see how to use it.

We can't plan for every possible incorrect usage, but is there any harm in adding a single line to the error messages?
I understand not calling `connect()` is improper usage of the library but IMO it's an easy mistake to make, especially since that worked in older library versions.
I will be more than happy to make the changes if you agree.

Either way, we really appreciate your help debugging this.

janiversen, 2023-08-19:
pull requests are welcome.