pymodbus-dev/pymodbus #1746: Empty result in REPL

raphapassini, 2023-08-26:
The documentation says that I should have a value in the `result` variable after successful running a command, which does not holds True.

```
➜ pymodbus.console tcp --host 0.0.0.0 --port 5020

----------------------------------------------------------------------------
__________          _____             .___  __________              .__
\______   \___.__. /     \   ____   __| _/  \______   \ ____ ______ |  |
 |     ___<   |  |/  \ /  \ /  _ \ / __ |    |       _// __ \\\____ \|  |
 |    |    \___  /    Y    (  <_> ) /_/ |    |    |   \  ___/|  |_> >  |__
 |____|    / ____\____|__  /\____/\____ | /\ |____|_  /\___  >   __/|____/
           \/            \/            \/ \/        \/     \/|__|
                                        v1.3.0 - 3.4.1
----------------------------------------------------------------------------

> client.connect
true

> client.read_holding_registers count=2 address=4 slave=1
{
    "registers": [
        9,
        27177
    ]
}

> result.data
> result.raw
```

janiversen, 2023-08-26:
@dhoomakethu you are the local specialist, any comments ?

@raphapassini which version are you using, on what system ?

dhoomakethu, 2023-08-26:
I will take a look the coming week, may be something has changed in the latest version. Thanks

janiversen, 2023-08-26:
I just verified the problem on current dev, seems result is not assigned.

The result of the request is correctly shown in the display, but result is empty.

janiversen, 2023-08-26:
you cannot run "result.decode" without parameters:

from documentation:
```
> result.decode word_order=little byte_order=little formatters=float64
28.17

>
```

dhoomakethu, 2023-08-26:
@raphapassini  Please check in this branch `repl-result-decoder`

janiversen, 2023-08-26:
@dhoomakethu thanks that seems to solve the problem.

janiversen, 2023-08-26:
Fixed on dev.

commit 126b5af2 (2023-08-26): Fix #1746 . Return missing result (#1748)
closed 2023-08-26 by janiversen