pymodbus-dev/pymodbus #1749: UNIT=1 does not work

gerontec, 2023-08-28:
- Python: 3.9
- OS: Linux pi3 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
- Pymodbus: 3.4.1

```
client = ModbusClient(method='rtu', port=SERIAL, startbits=1, stopbits=1, timeout=2, baudrate=BAUD, parity='N',errorcheck="crc")
result = client.read_holding_registers(0x0010, 2, unit=1)

```
Result is
 DEBUG:pymodbus.logging:SEND: 0x0 0x3 0x0 0x10 0x0 0x2 0xc4 0x1f
but should be:
DEBUG:pymodbus.logging:SEND: **0x1** 0x3 0x0 0x10 0x0 0x2 0xc4 0x1f
right?

### Versions

- Python:
- OS:
- Pymodbus:
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config()
```

janiversen, 2023-08-28:
Please read the documentation, it is not supposed to work.

closed 2023-08-28 by janiversen

gerontec, 2023-08-28:
Thx , with the correct parameter slave=1 it works now...

janiversen, 2023-08-28:
It helps reading the documentation before filing an issue :-)