pymodbus-dev/pymodbus #1755: Best Performance for Reading Arbitrary Register Address

AndrewStuhr, 2023-08-30:
Do you have any recommendations for reading multiple registers spaced out in an arbitrary/sporadic manner?  For example, the following holding register addresses (with offset 40,000):

(1, 2, 55, 76, 1001, 1009, 1499, 5000, 5001, etc..)   (arbitrary numbers, could be anything)

While individual registers could be read, that seems quite slow.  My understanding is there's a 125 limit on how many successive registers can be read at once, so maybe grouping read requests could be helpful.

Thanks for your work on this library!

janiversen, 2023-08-30:
The Modbus protocol only support <address> + <count>, so it is hard to do something real efficient.

What you could do (I used that in a similar project), is to sort the list, and then group addresses that are adjacent or with 1-2 registers in between and read that in one go. This is more efficient on a serial line, with wait times etc, on a socket there normally are no reason to do that.

Most devices I know implement a loop over count, and that is typically the big delay (again unless you use a slow serial line).

Hope that helps a bit, thanks for your nice words.

Ohh while I remember use the async client it performs a couple of factors better than the sync client.

AndrewStuhr, 2023-08-31:
Thanks a bunch!

I tried testing the async performance increase using _examples/client_performance.py_ , but the performance ended up being the same as without async.  The modbus server I read from is hosted on a Siemens PLC - maybe the issue is with that server's handling of requests?

If you can think of something I'm doing wrong with async, please let me know!

Here's the code is used (replaced serial with tcp, decreased loop count from 1000 to 10):
`
#!/usr/bin/env python3
"""Test performance of client: sync vs. async

This example show how much faster the async version is.

example run:

(pymodbus) jan@MacMini examples % ./client_performance.py
--- Testing sync client v3.4.1
running 1000 call (each 10 registers), took 114.10 seconds
Averages 114.10 ms pr call and 11.41 ms pr register.
--- Testing async client v3.4.1
running 1000 call (each 10 registers), took 0.33 seconds
Averages 0.33 ms pr call and 0.03 ms pr register.
"""
import asyncio
import time

from pymodbus.client import AsyncModbusTcpClient, ModbusTcpClient
from pymodbus.transaction import (
    #    ModbusAsciiFramer,
    #    ModbusBinaryFramer,
    ModbusRtuFramer,
    ModbusSocketFramer,
    ModbusTlsFramer,
)

LOOP_COUNT = 10
REGISTER_COUNT = 10
ip = '192.168.100.44'
port = 502

def run_sync_client_test():
    """Run sync client."""
    print("--- Testing sync client v3.4.1")
    client = ModbusTcpClient(
        ip,
        framer=ModbusSocketFramer,
        port=port,
    )
    client.connect()
    assert client.connected

    start_time = time.time()
    for _i in range(LOOP_COUNT):
        rr = client.read_input_registers(1, REGISTER_COUNT, slave=1)
        if rr.isError():
            print(f"Received Modbus library error({rr})")
            break
    client.close()
    run_time = time.time() - start_time
    avg_call = (run_time / LOOP_COUNT) * 1000
    avg_register = avg_call / REGISTER_COUNT
    print(
        f"running {LOOP_COUNT} call (each {REGISTER_COUNT} registers), took {run_time:.2f} seconds"
    )
    print(f"Averages {avg_call:.2f} ms pr call and {avg_register:.2f} ms pr register.")

async def run_async_client_test():
    """Run async client."""
    print("--- Testing async client v3.4.1")
    client = AsyncModbusTcpClient(
        ip,
        framer=ModbusSocketFramer,
        port=port,
    )
    await client.connect()
    assert client.connected

    start_time = time.time()
    for _i in range(LOOP_COUNT):
        rr = await client.read_input_registers(1, REGISTER_COUNT, slave=1)
        if rr.isError():
            print(f"Received Modbus library error({rr})")
            break
    client.close()
    run_time = time.time() - start_time
    avg_call = (run_time / LOOP_COUNT) * 1000
    avg_register = avg_call / REGISTER_COUNT
    print(
        f"running {LOOP_COUNT} call (each {REGISTER_COUNT} registers), took {run_time:.2f} seconds"
    )
    print(f"Averages {avg_call:.2f} ms pr call and {avg_register:.2f} ms pr register.")

if __name__ == "__main__":
    run_sync_client_test()
    asyncio.run(run_async_client_test())
`

And here's the results:
`
--- Testing sync client v3.4.1
running 10 call (each 10 registers), took 6.67 seconds
Averages 667.12 ms pr call and 66.71 ms pr register.
--- Testing async client v3.4.1
running 10 call (each 10 registers), took 6.79 seconds
Averages 679.13 ms pr call and 67.91 ms pr register.
`

closed 2023-09-04 by janiversen