pymodbus-dev/pymodbus #1757: Reading/writing non-integer values from/to registers

michK, 2023-08-31:
Hi.

I am able to read and write to coils and registers with the following snippets:

```
from pymodbus.client import ModbusTcpClient

# Turn on/off coil
client = ModbusTcpClient('127.0.0.1')
client.connect()
client.write_coil(0x4000, False)
client.close()

# Read coil
client = ModbusTcpClient('127.0.0.1')
client.connect()
result = client.read_coils(0x4000,1)
print(result.bits[0])
client.close()

# Write to integer register
client = ModbusTcpClient('127.0.0.1')
client.connect()
client.write_register(0x0000, 33)
client.close()

# Read from integer register
client = ModbusTcpClient('127.0.0.1')
client.connect()
result = client.read_holding_registers(0x0000)
print(result.registers[0])
client.close()
```

However `write_register` requires input to be `int`.

What if I want to write a float or string to a register? Or read them from registers? I see the functions `convert_from_registers` and `convert_to_registers` which I think might be what I should use, but can't figure it out.

Thanks in advance, and thank you for sharing this very useful package.

janiversen, 2023-09-01:
Please read the documentation.

michK, 2023-09-01:
I tried but cannot figure it out, can you refer me to the specif section of
the documentation?

michK, 2023-09-01:
For anyone else stumbling onto this, I figured this out with the help of [this](https://stackoverflow.com/questions/30784965/pymodbus-read-write-floats-real) stackoverflow post.

To write/read 32 bit float to/from address 0x000:

```
from pymodbus.client import ModbusTcpClient
from pymodbus.payload import BinaryPayloadBuilder,BinaryPayloadDecoder
from pymodbus.constants import Endian

# Write float to register
client = ModbusTcpClient('127.0.0.1')
client.connect()
builder = BinaryPayloadBuilder(byteorder=Endian.BIG, wordorder=Endian.LITTLE)
builder.add_32bit_float(77.77)
payload = builder.build()
result  = client.write_registers(0x0000, payload, skip_encode=True)
client.close()

# Read float from register
client = ModbusTcpClient('127.0.0.1')
client.connect()
result = client.read_holding_registers(0x0000,2)
decoder = BinaryPayloadDecoder.fromRegisters(result.registers, Endian.BIG, wordorder=Endian.LITTLE)
print(decoder.decode_32bit_float())
client.close()
```

closed 2023-09-01 by michK