pymodbus-dev/pymodbus #1759: REPL: pymodbus.server (RTU) does not start

chkr-private, 2023-09-03:
### Versions

- Python: 3.11.4
- OS: Fedora 37
- Pymodbus: 3.5.0

### Pymodbus Specific

- Server: rtu

### Description
- starting pymodbus.server fails
```pymodbus.server run -s serial -p /dev/ttyUSB0 -f rtu```

### Current result
```
2023-09-03 20:49:24,704 ERROR logging:114 Error starting modbus server 'ModbusSerialServer' object has no attribute 'start'
```
- the problem got introduced by commit 3d71883da1c71b4b31d090582ead659a9fc58623 which removed the start() method of ModbusSerialServer
- in previous pymodbus versions: https://github.com/pymodbus-dev/pymodbus/blob/8de66a2cf046537cabe390f4dd0d57c3b127005c/pymodbus/server/async_io.py#L594
- the start method is still called in pymodbus/server/reactive/main.py in start_modbus_server https://github.com/pymodbus-dev/pymodbus/blob/d1af6910bb05b61b0a97e6679f58b0df4e3f58d2/pymodbus/server/reactive/main.py#L242 https://github.com/pymodbus-dev/pymodbus/blob/d1af6910bb05b61b0a97e6679f58b0df4e3f58d2/pymodbus/server/reactive/main.py#L250

janiversen, 2023-09-03:
Pull request is welcome.

janiversen, 2023-09-04:
The start method was empty so it should be safe to remove calls.

janiversen, 2023-09-06:
Solved.

closed 2023-09-06 by janiversen