pymodbus-dev/pymodbus #1761: Exceptions are returned instead of raised.

matteo-zanoni, 2023-09-04:
In various places exceptions that occur are returned as return value instaed of being raised as exceptions:

https://github.com/pymodbus-dev/pymodbus/blob/d1af6910bb05b61b0a97e6679f58b0df4e3f58d2/pymodbus/transaction.py#L246C1-L252C27

janiversen, 2023-09-04:
Well behaving application test ModbusResponse.isError() so how can the application assume the request completed successfully. See our examples.

In general exceptions originating from the device are returned as ModbusResponse, and exceptions are only used for problems in the library (which in theory should never occur).

Your reference is a good example for a couple of reasons, we have received garble from the device
-so we tell the application (with isError) that something went wrong. Without having dived into the specific code, I assume that mixin packs the exc in a ModbusResponse if not that is a bug.
- and depending on the client parameters the library automatically retries.

matteo-zanoni, 2023-09-05:
The issue I encountered was, infact, that the exception returned in the example I linked was never wrapped in a `ModbusResponse`.

Also some of the confusion comes from the type hints and docstrings of functions like `read_holding_registers`:

```python
    def read_holding_registers(
        self, address: int, count: int = 1, slave: int = 0, **kwargs: Any
    ) -> ModbusResponse:
        """Read holding registers (code 0x03).

        :param address: Start address to read from
        :param count: (optional) Number of coils to read
        :param slave: (optional) Modbus slave ID
        :param kwargs: (optional) Experimental parameters.
        :raises ModbusException:
        """
    ...
```

Here it seems like `ModbusException` could be raised from the function but what happens is that (at least some times, not sure if everytime) the exception is actually returned.

If the intention is to always return and never raise then I would suggest (as @janiversen mentioned) wrapping the exception in a `ModbusResponse` and remove the `:raises ModbusException:` from the docstrings.

janiversen, 2023-09-05:
I tried to explain that we both use isError and exception, depending on the type of error.

Generally speaking:
- isError is returned when it is something the app should handle as part of the normal sequence.
- Exception is returned for an abnormal situation, mostly from something going wrong within the library itself. This is typically something the app cannot handle without a reset of the client.

janiversen, 2023-09-05:
The execute you refer to handled in https://github.com/pymodbus/client/base.py self._build_response,

matteo-zanoni, 2023-09-06:
Jut to be sure we are on the same page I developed the following test.
Running a modbus server simulator on my PC (so I can connect to it via Modbus TCP).
I modified in _pymodbus/framer/socket_framer.py_ the `ModbusSocketFramer._process` method to make it raise `ModbusIOException` always, so from:

```python
    def _process(self, callback, tid, error=False):
        """Process incoming packets irrespective error condition."""

        data = self._buffer if error else self.getFrame()
        if (result := self.decoder.decode(data)) is None:
            self.resetFrame()
            raise ModbusIOException("Unable to decode request")
        if error and result.function_code < 0x80:
            raise InvalidMessageReceivedException(result)
        self.populateResult(result)
        self.advanceFrame()
        if tid and tid != result.transaction_id:
            self.resetFrame()
        else:
            callback(result)  # defer or push to a thread?
```

I updated to:

```python
    def _process(self, callback, tid, error=False):
        """Process incoming packets irrespective error condition."""
        raise ModbusIOException("Unable to decode request")
```

Now I created a test program:

```python
from pymodbus.client.tcp import ModbusTcpClient
from pymodbus.pdu import ModbusResponse

c = ModbusTcpClient("172.30.64.1", port=1502)
try:
    c.connect()
    resp = c.read_holding_registers(1)
    print(f"Response is of type {type(resp)}")
    print(f"Is response a `ModbusResponse`? -> {isinstance(resp, ModbusResponse)}")
finally:
    c.close()
```

The output is:

```
Response is of type <class 'pymodbus.exceptions.ModbusIOException'>
Is response a `ModbusResponse`? -> False
```

So it looks to me like the `ModbusIOException` is never wrapped in a `ModbusResponse`.
How would you suggest wrapping the `ModbusIOException` in a `ModbusResponse`?
The `ModbusIOException` has a message (that is not present in `ModbusResponse` and also it has a `isError` method that always returns `False` (so it is different than the one from `ModbusResponse`.

There is a `ExceptionResponse` in _pdu.py_ but it looks to me like it is meant to represent an Exception returned from the modbus protocol (not our case).

I would appreciate any help!

janiversen, 2023-09-06:
Why wrap it, simply return a ModbusResponse, with error set.

matteo-zanoni, 2023-09-06:
Not sure `ModbusRresponse` is meant to hold the information of `ModbusIOException`...

Firstly `ModbusResponse` requires a `function_code` and in thi ssituation we were not able to process the response from the modbus server/slave so we have no function code. We could use the request's function code but then `ModbusResponse.isError` would return `False`.

Also returning a `ModbusResponse` would mean the end-user looses information about what error occurred.

janiversen, 2023-09-06:
do as you please...the library is working as it is, but we are always open to suggestions in form of Pull Requests. This is a "doers project", those who "do" rule.

Halkcyon, 2023-09-08:
Just my 2c, I think it's a pretty big anti-pattern to return a `ModbusIOException` as a value, especially when you are inheriting from `Exception` as a base class.

janiversen, 2023-09-08:
You are welcome, but as I said if you want changes make them instead of just complaining!

Do not forget we are all volunteers, and do what we want todo, not what others tell us to do.

janiversen, 2023-09-08:
Closing this as it is not a bug, possibly an enhancement, but that should be done as a pull request.

closed 2023-09-08 by janiversen

janiversen, 2023-09-08:
@Halkcyon did you actually look at the code ? Is it really an anti pattern to return different classes, because then the library will never work, we have many flavours of ModbusResponse depending on e.g. the response contains bits or registers.... ModbusIOException is one of those flavours.

matteo-zanoni, 2023-09-08:
@janiversen actually `ModbusIOException` is not a flavour of `ModbusResponse` (it does not have `registers` nor `bits`), and that is the issue raised.

janiversen, 2023-09-08:
no it does not...nor do all ModbusResponse flavours, some only have bits others only registers and some none of those.

BUT all have isError() which you are supposed to check.

janiversen, 2023-09-08:
All exception are a flavour of ModbusResponse, see exceptions.py:
```
    def isError(self):
        """Error"""
        return True
```

Apart from that it would be just as wrong:
- if ModbusResponse had .registers, when requesting coils
- if ModbusResponse had .bits, when requesting registers
- if ModbusResponse had .bits or .registers when requesting diagnostic

So as you can see a response can contain .bits or .registers or None and are all valid....important is that all responses contain .isError(), so the app can test the response with a simple "if".

But as I now have said many times, do not complain, suggest a change in the form of a pull request if you want something changed, but do not expect others to do that for you.

matteo-zanoni, 2023-09-08:
@janiversen sorry but what you are stating is just not true... I understand your design choice but the implementation does not reflect it.

Infact `ModbusResponse` **always** has `.registres` and `.bits`, if they are not used they are set to an empty list. This is the initializer of `ModbusResponse`:

```python
class ModbusResponse(ModbusPdu):
    def __init__(self, slave=0, **kwargs):
        """Proxy the lower level initializer.

        :param slave: Modbus slave slave ID

        """
        super().__init__(slave, **kwargs)
        self.bits = []
        self.registers = []
```

As you can see `bits` and `registers` always exist.

On the other hand trying to access the `registers` (or `bits`) of a  `ModbusIOException` will raise an `AttributeError`. This is why `ModbusIOException` **is not** a `ModbusResponse` (you can check it yourself with `isinstance(ModbusIOException(), ModbusResponse)`).

If your intention was to consider type hint the `read_holding_registers` (and similar) function to say that they return an object that is guaranteed to have a `isError` method than maybe a better approach would be something like this (using `typing.Protocol`):

```python
from typing import Protocol
from abc import abstractmethod

class ModbusResponseType(Protocol):
    @abstractmethod
    def isError(self) -> bool:
        raise NotImplementedError
```

This way the only thing that is guaranteed would be that `isError` is present.

janiversen, 2023-09-08:
Well you have made a PR, lets see how that works out. I am still looking forward to see the difference in the app code before and after your PR, a real difference in the code will surely convince me.

The main thing here is you should not use neither .register nor .bits if isError() returns true.