pymodbus-dev/pymodbus #1785: StartSerialServer not anymore maintain ?

yfourel-vossloh, 2023-09-21:
### Versions
Python: 3.8.10
OS: Ubuntu 20.04
Pymodbus: 3.5.2

### Description
I have a board with a master rs485 com using modbus. I would like to simulate device on my PC, that's why I try to use pymodbus lib to do a slave simuator.
Before using pymodbus, I do tests using a simple serial com with "serial.Serial" and printing "ser.read(15)" and it works I can see bytes that the master send me.
My problem is that "StartSerialServer" returns to me "TypeError: StartSerialServer() takes 0 positional arguments but 1 was given" when I pass it arguments. I don't get how it is possible because it needs arguments.

### Code

import serial
import serial.tools.list_ports as port_list
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.transaction import ModbusRtuFramer
from pymodbus.server import StartSerialServer
from pymodbus.datastore import ModbusSequentialDataBlock, ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

ports = list(port_list.comports())
com_port = ports[4].device
baud_rate = 115200

block = ModbusSequentialDataBlock(0x00, [1]*0xff)
store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
context = ModbusServerContext(slaves=store, single=True)
identity = ModbusDeviceIdentification()

print("before StartSerialServer")
server = StartSerialServer(context, identity=identity, port=com_port, framer=ModbusRtuFramer, baudrate=baud_rate, timeout=0.5)
print("after StartSerialServer")

try:

        print("server start")
        server.start()
        print("server online")
        while True:

            continue

except:

        server.stop()
        print("server stop")

janiversen, 2023-09-21:
StartSerialServer is supported, but the call have changed (as noted in API_CHANGES.rst), please read the documentation. 

You are using a positional parameter.

yfourel-vossloh, 2023-09-21:
Thanks for your answer. I read doc for 3.5.2 about StartSerialServer (https://pymodbus.readthedocs.io/en/latest/source/library/server.html#pymodbus.server.StartSerialServer) but the only example is about "StartAsyncSimulatorServer".

I changed args and now I don't have args error, but I never go out the function. I am in synchro mode so I guess I wait start ok but never ok.

```
server = StartSerialServer(
            context=context,
            identity=identity,
            timeout=1,
            port=ports[4].device,
            framer=ModbusRtuFramer,
            stopbits=1,
            bytesize=8,
            parity="E",
            baudrate=115200,
            strict=True,
        )
```

closed 2023-09-21 by yfourel-vossloh